pvvx/ZigbeeTLc #1: Bricked device

wildekek, 2023-11-13:
Looks like my LYWSD03MMC was bricked after flashing the firmware. The display does not turn on after flashing and can't find it in my zigbee network.

Things I tried:

Logs:
```
3:48:30 PM: Searching for devices
3:48:42 PM: NotFoundError: User cancelled the requestDevice() chooser.
3:48:42 PM: Reconnect 1 from 5
3:49:07 PM: Searching for devices
3:49:35 PM: Connecting to: Climate_1
3:50:00 PM: Hardware Revision String: B1.4
3:50:00 PM: Software Revision String: V4.3
3:50:00 PM: Firmware Revision String: github.com/pvvx
3:50:00 PM: Detected custom Firmware
3:50:11 PM: Hardware Version: LYWSD03MMC B1.4, Software Version: 4.3, Sensor: SHTC3 (SHTV3)
3:50:11 PM: Custom config HEX string: 55031002002804a9313180b4
3:50:31 PM: Load firmware file 'Z03MMC_v0102.bin'...
3:50:32 PM: File: Z03MMC_v0102.bin
3:50:32 PM: File size: 128900 bytes
3:50:32 PM: Count: 8057
3:50:36 PM: Start DFU
3:51:07 PM: Update done after 30.577 seconds
3:51:08 PM: Disconnected.
```

wildekek, 2023-11-13:
Ok, cool. Will try to flash it over UART and will re-flash with 0.1.0.3. Thanks for your work!

wildekek, 2023-11-14:
I [hooked it up to an FTDI ](https://github.com/atc1441/ATC_MiThermometer/blob/master/Mi_SWS_Connection.jpg) and used your [USB-COM Flash Writer ](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html). It gave no errors when erasing and writing ATC_v44.bin. Also, the "Soft Reset MCU" worked fine. It still does not show anything on the display.
Any ideas on what I could try to unbrick it?

pvvx, 2023-11-14:
A very strange situation.
"USB-COM Flash Writer" has no feedback and cannot verify the write. You may have to repeat the flashing...

closed 2023-11-25 by pvvx