pvvx/ZigbeeTLc #2: Не обновляется в z2m по ota

Gita1985, 2023-11-15:
[image]

pvvx, 2023-11-15:
Что и во что не обновляется?

Gita1985, 2023-11-16:
Прошил по вашей инструкции LYWSD03MMC в zigbee. В z2m отображается и работает, но не обновляется по OTA или так и должно быть?

pvvx, 2023-11-16:
Во что он должен обновиться?
Я не публиковал в общие списки zigbee прошивок, пока они в Beta стадии.
Скопируйте новые файлы вручную в каталог для обновления.
Я использую ZHA.
Если какие-то сложности, то берете файлы из [zigbee_ota to ble](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) и термометр перешивается обратно в BLE. Далее опять накатываете любую версию Zigbee, хоть со всеми новыми идентификаторами.
Одна проблема - после прошивки zigbee в ble, требуется стереть файлы fw и перезагрузить HA. 
Иначе он опять обновит всё в BLE, т.к. где-то помнит прошивки до полного перезапуска...

Gita1985, 2023-11-16:
Ок. Спасибо! Т.е. в дальнейшем должно работать обновление? Сейчас проблем нет. Все работает

pvvx, 2023-11-16:
В ver 0.1.0.4 [серьезные исправления](https://github.com/devbis/z03mmc/issues/47#issuecomment-1813514345) по поводу передачи измерений.

Gita1985, 2023-11-16:
А как мне обновить? Есть инструкция обновления устройства, которое уже перешито в ZigBee? Заранее благодарю

pvvx, 2023-11-16:
С z2m не работал.

closed 2023-11-16 by Gita1985