pvvx/ZigbeeTLc #4: Cannot find device in TelinkMiFlasher.html after flashing with Zigbee 0.1.0.5

jeremysherriff, 2023-11-16:
Should I be able to still find the device via BT scan, or is this function lost?
I have tried removing battery, and shorting GND/RST pins - this results in ` - - - ` being displayed and allows to to pair to a Zigbee controller, but does not allow me to find the device using a BT scan.

How do I flash the next version of firmware once it is released, or return to factory or BLE-based custom firmware?  Is this only available via a USB connection now?

(Note I am not 100% sure of the Zigbee firmware version applied, Z2M shows firmware `3001-0105` with build date `20231116`)

nonflammable, 2023-11-16:
After changing the firmware to ZigBee, you lose the ability to communicate with the device via Bluetooth. You can restore the firmware via OTA back to support Bluetooth.
https://github.com/pvvx/ZigbeeTLc/issues/3#issuecomment-1815039269

If a device with fw 3001-0105 is visible in Z2M, it means that it has been uploaded correctly.
I am afraid that it is not possible to support ZigBee and Bluetooth at the same time.

jeremysherriff, 2023-11-16:
Thanks for the comment but I am not quite understanding.

> After changing the firmware to ZigBee, you lose the ability to communicate with the device via Bluetooth. You can restore the firmware via OTA (or USB) back to support Bluetooth. [#3 (comment)](https://github.com/pvvx/ZigbeeTLc/issues/3#issuecomment-1815039269)

All the links in this comment show the Telelink html page that uses BT, I'm not sure what you mean in order to

> This may be useful to apply OTA "manually": [devbis/z03mmc#32](https://github.com/devbis/z03mmc/issues/32)

I will need to search more about this, but yes it looks promising as a method.

> If a device with fw 3001-0105 is visible in Z2M, it means that it has been uploaded correctly. I am afraid that it is not possible to support ZigBee and Bluetooth at the same time.

Agreed, the firmware appears to have updated correctly.  But My Z2M shows no data for the temp/humidity, I simply get an lqi and nothing else.

jeremysherriff, 2023-11-16:
Initially it looks good:
[image]

But no useful data:
[image]

State tab simply shows:
```
{
    "linkquality": 116
}
```

pvvx, 2023-11-16:
To restore to BLE, use the Zigbee OTA file with the same number from: https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

After running Zigbee OTA, be sure to delete this file or files from the update directories for Zigbee OTA and completely restart z2m/zha. Otherwise, with the new Zigbee firmware, this update in BLE will work again.

To update Zigbee, use the following files:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin

jeremysherriff, 2023-11-16:
I believe there is nothing wrong with the firmware, something goes wrong with the inspection when the sensor joins the zigbee network.  It worked first time but then stopped sending updates immediately, and removal and re-join now results in messages regarding invalid format.  Before the move to zigbee, the battery was reported as 100% (99% on display, 100% via BLE to Home Assistant) so I don't think it is that.

pvvx, 2023-11-16:
[image]

jeremysherriff, 2023-11-16:
Sorry pvvx but I have no idea what you're trying to show me there :)

I've not been successful with pushing the BLE firmware OTA using Zigbee2MQTT - in every attempt, I receive the message `Device didn't respond to OTA request`. It's sitting right beside the coordinator (and because of that, is now showing proper temp/humidity data, proving my original issues were related to poor reception).

Z2M uses a local database of the OTA files, and attempts to push an arbitrary FW appear to first require the database to recognise that there is a later firmware available to push.  If I can find a way to simply send the OTA file to the device without having to trick the Z2M database first, it might be successful.  I haven't found a way to do that yet, and I'm not sure Z2M provides such a method or if I need to find an alternate tool.

(Side note: I'm starting to wonder if some of the issue relates to the `-` in the current firmware file version as Z2M requires an integer.)

pvvx, 2023-11-17:
While the firmware options are in beta testing, I will not post them in the [general lists of automatic updates](https://github.com/Koenkk/zigbee-OTA) for all Zigbee devices.

jeremysherriff, 2023-11-17:
I understand.  I have made progress over the past few hours and now have Z2M showing an available update for the device.  (Unfortunately Z2M does not offer a mechanism to upload an arbitrary file to a device, which would make this process much easier..)

I am now encountering two different errors that are blocking the OTA mechanism - one is a generic timeout which I believe is somewhat expected given the nature of Zigbee and this device (LYWSD03MMC). The other appears to be a coding error with an unhandled exception in the node.js components of Z2M ("Attempt to access memory outside buffer bounds") - I'm just gathering info now to open an issue in the Z2M repo.

Closing as this is definitely not a problem with any components from this repo.

closed 2023-11-17 by jeremysherriff

jeremysherriff, 2023-11-18:
@pvvx I have resolved my problems with returning to BLE using Z2M.

I have upgraded my LYWSD03MMC to v0106.  Is there anything specific you want as feedback for the beta firmware?

pvvx, 2023-11-18:
Version 0.1.0.7 is currently being tested.
There are many changes in the firmware code.
How the preliminary test goes, I’ll post it on GitHub later...