pvvx/ZigbeeTLc #5: Custom manufacturer code

devbis, 2023-11-17:
Can you please use a custom manufacturer's code for your firmwares while they are in beta as they are incompatible with z03mmc in some ways and it causes issues with upgrade.

https://github.com/pvvx/ZigbeeTLc/blob/master/src/version_cfg.h#L53

Feel free to use telink code or any different for other devices besides 03 image type.

pvvx, 2023-11-17:
Все коды вариантов прошивок давно согласованы со всеми известными писателями, использующими TelinkID.

https://github.com/slacky1965/watermeter_zed/issues/1#issuecomment-1815323125

Вариант [0203 с версией 9999](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) используется только для восстановления прошивки термометра Xiaomi LYWSD03MMC из вашей прошивки в BLE и никогда не будет публиковаться.

Аналогично ни одна моя beta версия не будет опубликована в официальные списки прошивок zigbee. https://github.com/Koenkk/zigbee-OTA

И установите идентификатор для [ATC_ota_400000](https://github.com/devbis/z03mmc/blob/master/assets/ATC_ota_40000.bin) отличающий её от типовых прошивок. С ней возникли вопросы о блокировке термометров https://github.com/pvvx/ATC_MiThermometer у пользователей

devbis, 2023-11-17:
Да, окей, главное чтобы они не пытались обновляться через ОТА. 
В счетчике воды будет другой код производителя, мы обсудили с автором. 
Используя код телинка сейчас вы обрекаете себя и других писателей прошивок на трудности с обновлениями.
Я пожалел, что использовал его и теперь возникли трудности, надеюсь, разрешимые.

closed 2023-11-17 by devbis

pvvx, 2023-11-17:
Т.е. занять код другого производителя? Это может привести к аналогичным проблемам...

devbis, 2023-11-17:
Занять код производителя из далёкого будущего. Сейчас максимум это 0x1740, т.е. 0x4000-0xffff выглядит безопасно на ближайшие 10 лет.

pvvx, 2023-11-17:
0xDEAD :)