pvvx/ZigbeeTLc #7: OTA update in ZHA

pvvx, 2023-11-19:
## OTA update in ZHA.

The "configuration.yaml" file must have a directory assigned to the OTA files and the "ZHA Toolkit" installed.

For example, the directory name will be: “/zigpy_ota”
```
zha:
  zigpy_config:
    ota:
      otau_directory: /zigpy_ota
```
Place the "*.zigbee files" from [bin](https://github.com/pvvx/ZigbeeTLc/tree/master/bin) into this directory (“/zigpy_ota”).

Restart ZHA (preferably the entire Home Assistant).

Go to “Developer tools” -> “SERVICES” and select “ZHA Toolkit: Trigger Device's Firmware Update”.

[image]

Select “Device Reference” and click “CALL SERVICE”.

After some time, 15..20 minutes, the firmware number will change:

[image]

---

To restore to BLE, use the Zigbee OTA file with the same number device from: https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

After running Zigbee OTA, be sure to delete this file or files from the update directories for Zigbee OTA and completely restart z2m/zha. Otherwise, with the new Zigbee firmware, this update in BLE will work again.

bs76, 2023-11-25:
Tried to follow restoring BLE but stumbled on these issues:
- .zigbee file is not identified as OTA firmware:
```
2023-11-25 14:11:43.476 DEBUG (SyncWorker_1) [zigpy.ota.provider] File '/config/zigpy_ota/1141-020a-99993001-ATC_v46.zigbee' doesn't appear to be a OTA image
    return HueSBLOTAImage.deserialize(data)
    header, remaining_data = OTAImageHeader.deserialize(data)
ValueError: Wrong magic number for OTA Image: 168430090
    return OTAImage.deserialize(data)
    hdr, data = OTAImageHeader.deserialize(data)
```
- as follow up the firmware is not found:
```

pvvx, 2023-11-25:
Download the file correctly.
Not as “drop”, but as “raw”.

bs76, 2023-11-25:
D'oh ! Embarrasing mistake on my side. Thank you, succesfully reflashed

breaver, 2023-11-26:
> ## OTA update in ZHA.
> The "configuration.yaml" file must have a directory assigned to the OTA files and the "ZHA Toolkit" installed.
>
> For example, the directory name will be: “/zigpy_ota”
>
> ``` [...]

Do you have such an example for zigbee2mqtt?

pvvx, 2023-11-27:
> Do you have such an walkthrough for zigbee2mqtt?

https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#using-the-ikea-tradfri-test-server

breaver, 2023-11-28:
I followed the procedure but still not able to do the OTA update back to BLE. Z2M keeps telling me 
`Failed to check if update available for 'TF Sensor ' (No image available for imageType '522')`

- Used the raw file [1141-020a-99993001-ATC_v46.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v46.zigbee)
- added the local index file and placed everything into the root folder of z2m
- Any hint?

CLARENNE-Q, 2023-12-23:
Hello,

If you have HomeAssistant OS, the configuration file must be like this:

```
 zha:
   zigpy_config:
     ota:
       otau_directory: ./zigpy_ota
```

pvvx, 2023-12-23:
https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#enabling-ota-updates

onurunaldi, 2023-12-23:
> Hello,
> 
> If you have HomeAssistant OS, the configuration file must be like this:
> 
> ```
>       otau_directory: ./zigpy_ota
> ```

This solved my problem and this trick is not written anywhere. Thanks so much!

rwronski, 2023-12-28:
Unfortunately, OTA in zigbee2mqtt doesn't work for me either (HA OS). I've followed the manual, created override OTA index, tried many things, but it just doesn't work. If anyone has any other experience, please share.

rwronski, 2023-12-28:
Ok I've managed to make a progress - you don't add 'zigbee_ota_override_index_location' to the configuration.yaml file manually, but go to 'Settings' -> 'OTA Updates' -> fill in the 'OTA index override file name' field.  And suddenly z2m sees it, even though the configuration.yaml looks the same as if I'd edited it manually.

Neo2SHYAlien, 2023-12-29:
@rwronski try [this guide](https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/). Personally I didn't validate it

khayamgondal, 2024-02-10:
@pvvx where is the BLE .zigbee file for tuya TS0201?

djbuldog, 2024-02-17:
Hi, I have LYWSD03MMC with devbis fw 1.1.0 and I would like to migrate to ZigbeTLc fw or back to BT. I followed guides for Z2M and ZHA but without success. For example ZHA.. I added following to configuration.yaml:

```
zha:
  zigpy_config:
    ota:
      otau_directory: ./zigpy_ota

zha_toolkit:
```
the folder /root/config/zigpy_ota contains:
```
-rw-r--r--    1 root     root        128034 Feb 17 00:25 1141-0203-09993001-Z03MMC_v0120.zigbee
-rw-r--r--    1 root     root         82674 Feb 17 00:29 1141-0203-99993001-ATC_v46.zigbee
```
I called ZHA Toolkit: Trigger Device's Firmware Update service. The log contains:
```
2024-02-17 00:50:44.098 INFO (MainThread) [custom_components.zha_toolkit] Running ZHA Toolkit service: <ServiceCall zha_toolkit.ota_notify (c:01HPT58Q42H5KE2NXRDAPHC3HH): ieee=update.xiaomi_lywsd03mmc_z_firmware>
2024-02-17 00:50:44.917 DEBUG (MainThread) [zigpy.ota.provider] Inovelli: Finished downloading firmware update list
2024-02-17 00:50:45.101 DEBUG (MainThread) [zigpy.ota.provider] Ledvance: Finished downloading firmware update list
2024-02-17 00:50:45.470 DEBUG (MainThread) [zigpy.ota.provider] Salus: Finished downloading firmware update list
2024-02-17 00:50:45.521 DEBUG (MainThread) [zigpy.ota.provider] Sonoff: Finished downloading firmware update list
2024-02-17 00:50:46.003 DEBUG (MainThread) [zigpy.ota.provider] ThirdReality: Finished downloading firmware update list
2024-02-17 00:50:49.907 DEBUG (MainThread) [zigpy.zcl] [0x8416:1:0x0019] Decoded ZCL frame: Ota:QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4417, image_type=522, current_file_version=18886657)
2024-02-17 00:50:49.907 DEBUG (MainThread) [zigpy.zcl] [0x8416:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC-z': field_control=0, manufacturer_id=4417, image_type=522, current_file_version=18886657, hardware_version=None, model='LYWSD03MMC-z'
2024-02-17 00:50:49.907 DEBUG (MainThread) [zigpy.zcl] [0x8416:1:0x0019] No OTA image is available
```
there is one interesting message... I do not know what does it mean or if it is related to my issue... 
```
2024-02-17 00:34:06.142 INFO (MainThread) [zigpy.ota] OTA file providers are currently disabled
```
Core 2024.2.2
Supervisor 2024.01.1
Operating System 11.5
Frontend 20240207.1

It looks like the local OTA files are ignored... Can you see anything wrong? Any ideas what to try next?

robvanoostenrijk, 2024-02-17:
zigpy has disabled local providers because many people are breaking their devices with wrong OTA updates.

The commit that does that is here:
https://github.com/zigpy/zigpy/commit/6aeb8a6816585a4bda67e9ea1b125568c18fa702

You can still do OTA updates with the following yaml configuration:
(Note the new `allow_file_providers` key and its _value_)

    zha:
      zigpy_config:
        ota:
          allow_file_providers: "I understand I can *destroy* my devices by enabling OTA updates from files. Some OTA updates can be mistakenly applied to the wrong device, breaking it. I am consciously using this at my own risk."
          ikea_provider: true
          inovelli_provider: false
          ledvance_provider: false
          otau_directory: ./zigbee_ota
          salus_provider: false
          sonoff_provider: false
          thirdreality_provider: false

ckpt-martin, 2024-03-19:
After flashing the Zigbee firmware, is there a way to tell exactly which image was installed? I am unsure whether the image from pvvx or devbis were installed.
I have followed everything above, including the new allow_file_providers line and have download the ATC 4.7 files for restoring BT as raw and not blob. I uploaded both the 0203 and 020a images into the zippy_ota folder. Restarted HA, initiated the OTA image_notify command. But nothing happens. My devices always say firmware is up-to-date and the debug log says no compatible images found.
My devices are on Firmware 0x01203001. I just have one that constantly loses connectivity. So, I wanted to restore the BT image to start over, just in case something went wrong during flashing. If that doesn't help, then I assume a faulty device. In which case I would also have to restore BT image to return it.
Is there something else that needs to be done to allow ZHA to detect the correct firmware file under the zigpy_ota folder?

pvvx, 2024-03-19:
[Zigbee->BLE Files ](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) 
https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0203-99993001-ATC_v47.zigbee
https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v47.zigbee
```
zha:
  zigpy_config:
    ota:
      allow_file_providers: "I understand I can *destroy* my devices by enabling OTA updates from files. Some OTA updates can be mistakenly applied to the wrong device, breaking it. I am consciously using this at my own risk."
      otau_directory: ./zigbee_ota
```

ckpt-martin, 2024-03-19:
> [#7 (comment)](https://github.com/pvvx/ZigbeeTLc/issues/7#issue-2000695703)
> 
> [Zigbee->BLE Files ](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0203-99993001-ATC_v47.zigbee https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v47.zigbee
> 
> ```
> zha:
>   zigpy_config:
>     ota:
>       allow_file_providers: "I understand I can *destroy* my devices by enabling OTA updates from files. Some OTA updates can be mistakenly applied to the wrong device, breaking it. I am consciously using this at my own risk."
>       otau_directory: ./zigbee_ota

I had that setup. But you might want to modify your text above that you copied here. The directory that is referenced is "zigpy_ota", but in your example config you have "zigbee_ota". I suppose it doesn't matter as long as config and actual directory name are the same. ;-)
And maybe your links to images should be modified, replacing "blob" with "raw" to make sure people download the raw version.
I just tried the reconfigure option (after finally successfully pairing again) and it shows the same as your screenshots except the TemperatureMeasurement has an "X" next to the measured_value. :-(

maikvitesse, 2024-07-30:
Just got it working thanks to [this](https://github.com/home-assistant/core/pull/111159) and [this](https://github.com/zigpy/zigpy/pull/1340).

My working code now:

```yml
zigpy:
  ota:
    advanced_ota_dir: /config/zigpy_ota
    allow_advanced_ota_dir: I understand I can *destroy* my devices by enabling OTA
        updates from files. Some OTA updates can be mistakenly applied to the
        wrong device, breaking it. I am consciously using this at my own risk.
```

cfulton, 2024-08-28:
Newer working config
```yaml
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: advanced
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          path: /config/zigpy_ota
```

fmauNeko, 2024-09-22:
Even simpler
```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m
          url: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index.json
  ```

michi2005, 2024-09-25:
Dear all

I tried a merge of the last 2 options but without success.

```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          url: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index.json
```

But no new Firmware is shown....

sebgus, 2024-10-08:
How can I reflash the OTA image or downgrade it?

I've managed to upgrade the firmware of a TS0201-TZ3000 to '0x01233001' (reported in ZHA), but the temperature reading reports -327.7 degrees C and the humidity is 655.4%. So something has gone wrong somewhere...

Regressor, 2024-10-18:
> If you're using Zigbee2MQTT and want to update your devices from devbis firmware to ZigbeeTLc using Over-The-Air (OTA) updates, follow these simple steps.

This should be in main readme! Thank you!

satmandu, 2024-10-26:
@arthurkowalsky I'm trying this method to update from devbis firmware to ZigbeeTLc and I get this message:

`Failed to check if update available for '0xa4c1385f065ca6c7' (Attempt to access memory outside buffer bounds)`

Am I missing something?

```
Firmware build date
    20240204
Firmware version
    3001-1100
Manufacturer
    [Custom devices (DiY)](https://www.zigbee2mqtt.io/supported-devices/#v=Custom%20devices%20(DiY))
Model
    [LYWSD03MMC](https://www.zigbee2mqtt.io/devices/LYWSD03MMC.html#custom%2520devices%2520(diy)-lywsd03mmc)
```

I'm trying to update with [db15-0203-79993001-Z03MMC_v0122.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/devbis/db15-0203-79993001-Z03MMC_v0122.zigbee)

End of `/homeassistant/zigbee2mqtt/configuration.yaml`:
```yaml
ota:
  zigbee_ota_override_index_location: my_index.json
```

`/homeassistant/zigbee2mqtt/my_index.json`:
```yaml
[
    {
        "url": "db15-0203-79993001-Z03MMC_v0122.zigbee",
        "force": true
    }
]
```

Skol6, 2024-11-09:
> Even simpler - configuration.yaml
> 
> ```yaml
> zha:
>   zigpy_config:
>     ota:
>       extra_providers:
>         - type: z2m
>           url: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index.json
> ```

Hi Folks,
I tried above "simple" method to update my Xiaomi LYWSD03MMC temperature sensors. But actually I am failing.
Meanwhile I am trying for 5 hours now to load the update but I a have no clue where I am stuck.
The ZHA Toolkit was installed. I am using Home Assistant on a Proxmox Machine and not using Zigbee2MQTT.
Thanks a lot for your help.

[image]

Arachnid, 2024-12-21:
I have a TH03 from [this Aliexpress listing](https://www.aliexpress.com/item/1005007309529179.html?spm=a2g0o.order_detail.order_detail_item.2.72ccf19c4GpMOy). I'd like to OTA flash it to the custom firmware. I've confirmed the PCB matches the layout shown for TH03 devices, but it reports a manufacturerCode of 4742 and an imageType of 65535.

Further, the file listed in `bin` for upgrading Tuya/Mihome devices to the custom firmware, `1286-0202-10993607-TH03Zz_v0123.zigbee`, is not included in index.json.

I've downloaded index.json, and added the following entry:
```
    {
        "fileVersion": 268645896,
        "fileSize": 128146,
        "manufacturerCode": 4742,
        "imageType": 65535,
        "sha512": "81eaf9113318a887793f6c62bf8e4f07487c55248dcbcd7175c26c57514f556161007f706cd92ce3ab8e12999f2c3439a95b153833cdceeee4de054ad2fc6f31",
        "url": "https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1286-0202-10993607-TH03Zz_v0123.zigbee"
    }
```

I modified my configuration.yaml as described in the first entry here, and confirmed that ZHA is reading it. However, even after restarting home assistant, no firmware update shows as available for the device. Can anyone suggest why?

pvvx, 2024-12-21:
> it reports a manufacturerCode of 4742 and an imageType of 65535.

 "imageType": 514 (0x0202)

Arachnid, 2024-12-21:
> > it reports a manufacturerCode of 4742 and an imageType of 65535.
> 
> "imageType": 514 (0x0202)

Sorry, what do you mean by this?

> The transition from Tuya to other variants is not included in the general index file.
> The reason is that 90% of Tuya devices have the same IDs for OTA. And only the user can specify which device the OTA firmware is suitable for.
> The OTA Zigbee update menu will drop down several different devices, and users will, without thinking, select them all...
> [Koenkk/zigbee-OTA#620 (comment)](https://github.com/Koenkk/zigbee-OTA/issues/620#issuecomment-2515221969)
> [Koenkk/zigbee-OTA#620 (comment)](https://github.com/Koenkk/zigbee-OTA/issues/620#issuecomment-2526350630)

I see, that makes sense. Is there something wrong with my custom configuration that would cause it not to be recognized as an update for my device?

pvvx, 2024-12-21:
> Sorry, what do you mean by this?

It worked for me with this one

index.json:
```
[
    {
        "fileVersion": 278476295,
        "fileSize": 128146,
        "manufacturerCode": 4742,
        "imageType": 514,
        "sha512": "81eaf9113318a887793f6c62bf8e4f07487c55248dcbcd7175c26c57514f556161007f706cd92ce3ab8e12999f2c3439a95b153833cdceeee4de054ad2fc6f31",
"url": "https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1286-0202-10993607-TH03Zz_v0123.zigbee"
    }
]
```
But there is no guarantee that the firmware in your device has the same IDs.

Arachnid, 2024-12-21:
> But there is no guarantee that the firmware in your device has the same IDs.

Mine reports a manufacturerCode of 4742 and an imageType of 65535 when I check the cluster config, but this configuration doesn't show up as an available update.

pvvx, 2024-12-21:
ZHA does not have an interface that would show the device IDs to the user for OTA. IDs can only be found in the log when OTA occurs.

Arachnid, 2024-12-21:
Oh, I see. I had assumed that the values I retrieved from the ota cluster would match. Is there any way to get the relevant values if I don't already have an OTA to send?

pvvx, 2024-12-21:
There is, but there is no interface.

Arachnid, 2024-12-21:
Sorry, can you elaborate? How can I get the correct `imageType` value for my device?

pvvx, 2024-12-21:
The device provides identifiers when requesting the Zigbee OTA function.

Arachnid, 2024-12-21:
I see. I tried to use ZHA Toolkit as documented in your first post to prompt an OTA update. However, I get the error "sl_Status.ZIGBEE_DELIVERY_FAILED". In the system logs I also see "Could not get ota object for ota_update_images, try again", and a parsing error from zigpy trying to deserialize a package with too-short data:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/device.py", line 488, in packet_received
    hdr, args = endpoint.deserialize(packet.cluster_id, data)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/endpoint.py", line 228, in deserialize
    return cluster.deserialize(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/zcl/__init__.py", line 290, in deserialize
    response, data = command.schema.deserialize(data)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/types/struct.py", line 319, in deserialize
    value, data = field_type.deserialize(data)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/types/basic.py", line 874, in deserialize
    item, data = cls._item_type.deserialize(data)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/zcl/foundation.py", line 891, in deserialize
    r.direction, data = ReportingDirection.deserialize(data)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/types/basic.py", line 199, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```

None of this leaves me any closer to figuring out how to get the device's image type, however. Any advice you have to offer would be much appreciated.

pvvx, 2024-12-21:
No advice.
ZHA toolkit is not updated.
ZHA uses a new version and interfaces with zigpy.
Each new release of ZHA becomes more complex. It is possible that you will not find anything in the log in the new version of ZHA.

Arachnid, 2024-12-21:
Edit: I sent the following action, and suddenly an update shows up as available:
```
action: zha.issue_zigbee_cluster_command
data:
  ieee: [mac]
  endpoint_id: 1
  cluster_id: 25
  cluster_type: out
  command: 0
  command_type: client
  args:
    - 0
    - 100
```

At a guess, the device simply hadn't checked for an update since I updated the file?

pvvx, 2024-12-21:
> At a guess, the device simply hadn't checked for an update since I updated the file?

The check only occurs when the HA is completely rebooted...

Arachnid, 2024-12-21:
Unfortunately, the update still won't apply - it times out trying to OTA, even with me pressing the button to wake the device up.

pvvx, 2024-12-22:
OTA process interrupted?
My OTA took over an hour, with interruptions. I ran it several times and the update continued... The OTA program works very slowly on this device.

StancuFlorin, 2024-12-29:
I have 6 Xiaomi sensors (with model LYWSD03MMC-z) and I have succefully updated them using OTA with the config added bellow. I am using Home Assistant version `2024.12.5`

```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: ikea
        - type: z2m # Xiaomi LYWSD03MMC https://github.com/pvvx/ZigbeeTLc/issues/7#issuecomment-2365373959
          url: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index.json
          manufacturer_ids: [0x1141]
```

The installed version on these sensors was `0x01203001` and the new version is `0x01233001`. The update was pretty fast (around 30 minutes). It was more faster than the update for some Ikea sensors that took around 2h.

The controller that I am using is `SONOFF Zigbee 3.0 USB Dongle Plus V2`

I have another Xiaomi sensor with model LYWSD03MMC (without -z) with firmware `0x11003001`. It is possible that I flashed a different firmware when I transformed it from BT to Zigbee. For this one I got no update available. Still looking why.

pvvx, 2024-12-29:
> I also have another Xiaomi sensor, model LYWSD03MMC (without the -z), running firmware version 0x11003001.

https://github.com/devbis/z03mmc

StancuFlorin, 2024-12-29:
> > I also have another Xiaomi sensor, model LYWSD03MMC (without the -z), running firmware version 0x11003001.
> 
> https://github.com/devbis/z03mmc

There is any way to flash your firmware over the air?

pvvx, 2024-12-29:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis

StancuFlorin, 2024-12-29:
> https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis
> 
> https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota Zigbee -> BLE

How can I manually trigger the update? I’ve added the firmware file `db15-0203-79993001-Z03MMC_v0123.zigbee` to the `/config/zigpy_ota` directory, but my devbis LYWSD03MMC device shows "Firmware is up to date" in ZHA. Until now the firmware status was unknown.

pvvx, 2024-12-29:
Follow the ZHA documentation. Or ask the ZHA writers.

pvvx, 2024-12-29:
My current version of the mess for ZHA OTA is this:
```
zha:
  zigpy_config:
    ota:
      enabled: true
      # Download the index over HTTPS
      z2m_remote_index: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
      # Or, load the index offline
      z2m_local_index: /home/pvvx/zigpy_ota/index.json
      allow_advanced_ota_dir: I understand I can *destroy* my devices by enabling OTA
        updates from files. Some OTA updates can be mistakenly applied to the
        wrong device, breaking it. I am consciously using this at my own risk.
      advanced_ota_dir: /home/pvvx/zigpy_ota
      otau_directory: /home/pvvx/zigpy_ota
      path: /home/pvvx/zigpy_ota
      ikea_provider: false
      ledvance_provider: false
      salus_provider: false
      sonoff_provider: false
      inovelli_provider: false
      thirdreality_provider: false
```
I create a local index.json and put it in /home/pvvx/zigpy_ota/
After the update I change index.json to an empty one containing:
```
[
]
```

tuanhatran, 2025-01-26:
Hello @arthurkowalsky,
First thank you for you guide about how to flash devbis firmware to ZigbeeTLc
I could flash my Xiaomi LYWSD03MMC to version 0124 (latest firmware) from devbis 1100
However, on my Z2M, it still appear as "Custom devices (DiY)" and LYWSD03MMC.
Which is different from Xiaomi and LYWSD03MMC at some attribute : "comfort_display" for -z vs "show_smiley" and "display" vs "enable_display"
I also try to reflash from 0124 to ATC_v50 BLE, then flash it back using pvvx tool, it displays 0124 version, but still same issue with Custom devices (DiY)
Do you have any idea?
Thank you for your help
Best regards

pvvx, 2025-01-26:
> I also try to reflash from 0124 to ATC_v50 BLE, then flash it back using pvvx tool, it displays 0124 version, but still same issue with Custom devices (DiY)

The IEEE address of the devices does not change.
The Zigbee management program may not reread the device configuration, since the IEEE remains the same.
You need to delete and re-register the device.

tuanhatran, 2025-01-27:
> The IEEE address of the devices does not change. The Zigbee management program may not reread the device configuration, since the IEEE remains the same. You need to delete and re-register the device.

Thank you. I removed the device, restart Z2M, restart HA, then readd devices and it's correctly displayed now !

Sahta77, 2025-03-20:
@pvvx Can you show a simple example of your local index.json? I'm trying to flash a https://pvvx.github.io/TS0201_TZ3000/ device, but I'm struggling with setting everything up in order to have the update available. The big index.json from Koenkk is too complex for me to understand which fields I need to make sure are aligned in order for the update to be available.

pvvx, 2025-03-20:
> [@pvvx](https://github.com/pvvx) Can you show a simple example of your local index.json? I'm trying to flash a https://pvvx.github.io/TS0201_TZ3000/ device, but I'm struggling with setting everything up in order to have the update available. 

https://github.com/pvvx/ZigbeeTLc/tree/master/bin
https://github.com/pvvx/ZigbeeTLc/blob/master/bin/ts0201_tuya2z.json

robvanoostenrijk, 2025-03-21:
Example JSON for TS0201:
```
[
	{
		"url": "OTA/1141-0211-01243001-ZTS0201Z3000.zigbee"
	},
	{
		"url": "OTA/1141-0206-01243001-ZCGDK2.zigbee"
	}
]
```

pvvx, 2025-03-25:
The new version of ZHA is a mess again. OTA update does not work when specifying local files. In the device properties it shows the old version number, and in Firmware Update - the latest loaded version via OTA. Only the command to read the real version works and shows current data.

[image]

[image]

The ZHA writers take all measures to ensure that the user cannot update anything via Zigbee OTA.

pixolex, 2025-04-05:
Hello @pvvx 

I managed to update one of my xiaomi thermometer with this configuration.

I downloaded from the bin directory in github the .zeegbee file and the .Jason, and it worked.

zha:
  zigpy_config:
    ota:
      disable_default_providers: [ikea]  # Disable the default ikea provider
      extra_providers:
        - type: zigpy_local  # Look at `local_index.json` in the zigpy repo for an example
          index_file: ./otas/index.json
        - type: advanced  # !!! Please read the message below before enabling OTA updates from raw OTA files !!!
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          path: ./

pvvx, 2025-04-05:
But the menu will still show the old FW version number.

[image]

[image]

This was done in ZHA to make you completely confused. :)

pixolex, 2025-04-05:
[image]

I don't have the FW version in that window. Don't know why...

pvvx, 2025-04-05:
This comes out in ZHA as its writers wanted :)
Perhaps, the next time RF reboots, after some time it will show the version... But in the new version of ZHA it shows the old version number there...

[6 more replies]