pvvx/ZigbeeTLc #8: ZCGDK2 Completely Wrong Measurements

SZenglein, 2023-11-19:
The values are -39.18°C and 99.99% on the Zigbee Side. The LCD shows -39°C and 10.0%.

I tried your custom BT firmware and the measurements were correct. I reverted to it via Zigbee OTA and everything is correct again.

pvvx, 2023-11-19:
It is possible to obtain log data using BLE firmware for commands "02" and "05"?
[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

[image]
```
23:19:29: Settings 02 was send successful
23:19:30: Sensor: I2C addres 0x70, LCD driver: I2C addres 0x3E
```
[image]
```
23:18:12: Settings 05 was send successful
23:18:12: Sensor ID: 00000887
```

SZenglein, 2023-11-19:
``` 
22:02:46: Hardware Revision String: 2.1.0
22:02:46: Software Revision String: V4.5
22:02:46: Firmware Revision String: github.com/pvvx
22:02:46: Detected custom Firmware
22:02:48: Hardware Version: CGDK2 2.1.0, Software Version: 4.5, Sensor: SHTC3 (SHTV3)
22:02:48: Custom config HEX string: 55011000002804a9313186b4
22:03:42: Settings 02 was send successful
22:03:43: Sensor: I2C addres 0x70, LCD driver: I2C addres 0x3E
22:04:30: Settings 05 was send successful
22:04:31: Sensor ID: 00000000
```

pvvx, 2023-11-19:
Thank you.
Somewhere I made a mistake in the code for working with defective or fake sensors SHTV3.
https://github.com/pvvx/ATC_MiThermometer/issues/273
I'll try to fix it, but I don't have such thermometers. There is CGDK2 with normal SHTV3 sensors...

SZenglein, 2023-11-19:
I didn't read all of the linked issue, but that is some annoying hardware. I'm curious to see how the battery lifetime will be, as that appears to be an issue?

Anyway, let me know if I can provide you with more info or test a new image. I might also be able to attach via I2C.

pvvx, 2023-11-19:
It seems to me that a batch of sensors with an error in the crystal was used. This has been fixed in the official firmware.
From such sensors, data is read at address I2c 0x0, and control is performed at address 0x70 :)
This should not affect the battery

SZenglein, 2023-11-19:
Probably not related, but the missing capacitor on newer devices causes battery issues, right?

pvvx, 2023-11-19:
- 40% battery

closed 2023-11-25 by pvvx