pvvx/ZigbeeTLc #11: OTA update failed with reason: 'invalid image'

ikoz, 2023-11-20:
Some days ago I flashed a couple of LYWSD03MMC sensors from custom BTHome to the ZigbeeTLc firmware using TelinkMiFlasher.
 
I added the 2 devices in Z2M and they work fine so far - showing up as version 3001-0102

I am now trying to do OTA update via Z2M - either to latest 106 firmware or even back to BLE.

I've tried both firmware 1141-020a-01103001-Z03MMC.zigbee and 1141-020a-99993001-ATC_v46.zigbee

However in both cases Z2M shows update starting ..0% - and then immediatelly after it fails with "Update failed with reason: 'invalid image'"

```
 2023-11-20 14:31:56Got upgrade end request for '0xa4c138fda3d5689c': {"status":150,"manufacturerCode":4417,"imageType":522,"fileVersion":2576953345}
Debug 2023-11-20 14:31:56Update failed with reason: 'invalid image'
```

Not sure where to go from there - any way to ypdate via OTA or do I need to use USB cable?

nonflammable, 2023-11-20:
what size are the update files on your disk?

ikoz, 2023-11-20:
> what size are the update files on your disk? did you try to download this file: https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-020a-01103001-Z03MMC.zigbee size: 126274

yes that's the file, and the file size is correct on disk. 

```
$ sha1sum 1141-020a-01103001-Z03MMC.zigbee
ce1096a0f2a349b91649617fb8269bc4ca709d88  1141-020a-01103001-Z03MMC.zigbee
```

pvvx, 2023-11-20:
Before flashing the firmware, check the compliance of the “Manufacturer Code” and “Image Type” between the OTA file and those issued by the Zigbee device.

Sample: "1141-020a-01103001-Z03MMC.zigbee" 

| Manufacturer Code | Image Type | File Version | Stack Version | Name | Ext OTA |
| -- | -- | -- | -- | -- | -- |
| 1141 | 020a | 0110 | 3001 | Z03MMC | zigbee |
| 0x1141 | 0x02 - TLSR825x, 0x0a - Xiaomi LYWSD03MMC | App release 0.1, App build 1.0 | Zigbee v3.0, Release 0.1 |  Z03MMC | OTA |

ikoz, 2023-11-20:
Hmm not sure how to check these parameters. My device is 4417 (0x1141) and 522 image type (0x20A) as far as I can tell.

pvvx, 2023-11-20:
> I added the 2 devices in Z2M and they work fine so far - showing up as version 3001-0102

Beta version below 0.1.0.3 may have a [bug](https://github.com/pvvx/ZigbeeTLc/issues/1#issuecomment-1808749887) related to the not very common TLSR825x, which has an integrated flash memory die from Zbit Semiconductor.

ikoz, 2023-11-20:
> Beta version below 0.1.0.3 may have a [bug](https://github.com/pvvx/ZigbeeTLc/issues/1#issuecomment-1808749887) related to the not very common TLSR825x, which has an integrated flash memory die from [...]

Interesting. Both devices work fine and report temperatures fine, so I wouldn't consider them bricked though. Only the OTA doesn't work! I guess my only choice is to try to update using a cable.

pvvx, 2023-11-20:
TLSR825x chips with Flash from ZB (Zbit Semiconductor) with fw version lower than 0.1.0.3 cannot write Flash.
For this reason, OTA does not work. And any saving of the Zigbee configuration does not work.
Rewriting flash memory is only possible using any version of the hardware programmer.

closed 2023-11-25 by pvvx