pvvx/ZigbeeTLc #12: LYWSDCGQ/01ZM

aptem334, 2023-11-20:
Please add a sensor LYWSDCGQ/01ZM

aptem334, 2023-11-20:
MJ_HT_V1

aptem334, 2023-11-20:
[image]
[image]

pvvx, 2023-11-20:
This is a chip from [Nordic Semiconductor](https://www.nordicsemi.com/products/nrf51802). 
Only devices based on Telink TLSR825x series chips are discussed here.

nonflammable, 2023-11-20:
> Please add a sensor LYWSDCGQ/01ZM

Probably the nRF51802 chip does not support ZigBee

closed 2023-11-20 by pvvx

aptem334, 2023-11-28:
Как на счет того что бы перепаять микроконтроллер из SJWS01LM? Это сработает?

aptem334, 2023-11-28:
> This is a chip from [Nordic Semiconductor](https://www.nordicsemi.com/products/nrf51802).
> Only devices based on Telink TLSR825x series chips are discussed here.

How about resoldering the microcontroller from SJWS01LM? Will this work?

pvvx, 2023-11-28:
Everything is possible.
Example:  https://pvvx.github.io/CGDK2/CGDK2-2/