pvvx/ZigbeeTLc #13: Bricked CGDK2

martinrosenauer, 2023-11-20:
Hi, thanks a lot for doing these amazing hacks to these devices.

I tried to zigbee'fy a CGDK2 and had an idea it as the same procedure as for LYWSD03MMC. I flashed with ATC_ota_40000.bin which went fine but bricked the device. 

Any way to recover?

Thanks!

pvvx, 2023-11-20:
"ATC ota_40000.bin" is not supported and is not intended for CGDK2.

"ATC ota_40000.bin" is a file from https://github.com/devbis/z03mmc.
I don't know what "ATC ota_40000.bin" does.

martinrosenauer, 2023-11-20:
True, I mixed up the procedures and somehow figured it was needed to make the OTA update work. My mistake. Any suggestions besides dumping it in the trash?

pvvx, 2023-11-20:
> Any suggestions besides dumping it in the trash?

1. Try to find a new BLE device "ATC_xxx" and restore the firmware: [CGDK2_v46.bin](https://github.com/pvvx/ATC_MiThermometer/raw/master/CGDK2_v46.bin)
2. [The USB-COM adapter writes the firmware in explorer. Web version.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)

closed 2023-11-20 by pvvx