pvvx/ZigbeeTLc #9: OTA update in Z2M

PetrMa, 2023-11-20:
Hello,

I made a flash of my BT thermometer to Zigbee with your flasher on https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
Now I´am on version:
20231108
3001-0101
[image]

Is there any way how to make an OTA update to latest version in Z2M please?
I tried to download this file https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-020a-01103001-Z03MMC.zigbee and make manual OTA in Z2M but it ends all time with an error:
[image]

Thanks for your help

nonflammable, 2023-11-20:
You should download this file:
https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-020a-01103001-Z03MMC.zigbee
size: 126274

PetrMa, 2023-11-21:
@nonflammable I have the correct zigbee file but it ends wit the same error message

pvvx, 2023-11-21:
https://github.com/pvvx/ZigbeeTLc/issues/11#issuecomment-1819878948

PetrMa, 2023-11-21:
@pvvx OK, so the USB UART is the only one way at this time, right?

pvvx, 2023-11-21:
Yes

closed 2023-11-25 by pvvx