pvvx/ZigbeeTLc #14: TS0201 (WSD500A) identifying as different device in Z2M after flashing 0.1.1.1

slingel, 2023-11-23:
left original, right custom firmware. help appreciated.

[image]

pvvx, 2023-11-23:
[image]
[image]
[image]
[image]
[image]

slingel, 2023-11-23:
> Your program (integration) is wrong. Write to the program creators.
> 
> What kind of manufacturer is this with the name “_TZ3000_xr3htd96”? Where did “sensor with display” come from? Where is the manufacturer's number? Where is the Zigbee version number? What has the software degenerated to - it can no longer correctly read "Attributes for determining basic information about the device"? [image]
> 
> Cluster 0x0001: Basic ZCLVersion: 3 ApplicationVersion: 0 StackVersion: 2 HWVersion: 1 ManufacturerName: "Tuya" ModelIdentifier: "TS0201" DateCode: "20231121" PowerSource: 3 - Battery SWBuildID: "3001-0111" Cluster 0x0019: OTA Manufacturer ID: 0x1141 (Telink) or in the future there will be 0xD0E5 (DOES -> DIY) ...
> 
> [image] [image] [image] [image] [image]

pvvx, 2023-11-23:
Because Z2M is not user friendly, use the gag from the https://github.com/devbis/z03mmc/blob/master/converters/lywsd03mmc.js

change lines:
```
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: 'Tuya',
```
Or manually binding the required measurements (clusters).

slingel, 2023-11-25:
working fine now. ota update working too. thank you.

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['TS0201'],
    model: 'WSD500A',
    vendor: 'Tuya',
    description: 'Temperature & humidity sensor',
    fromZigbee: [fz.temperature, fz.humidity, fz.battery, fz.hvac_user_interface],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
        await reporting.temperature(endpoint, {min: 10, max: 300, change: 10});
        await reporting.humidity(endpoint, {min: 10, max: 300, change: 50});
        await reporting.batteryVoltage(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [
        e.temperature(), e.humidity(), e.battery(), e.battery_voltage(),
    ],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

[image]

closed 2023-11-25 by pvvx

khayamgondal, 2024-02-09:
@slingel I am facing the same issue. Where do I paste the code above to get it corrected? @pvvx

slingel, 2024-02-09:
in the zigbee2mqtt data folder create a converter named ts0201.js or something like that and then add it to the bottom of configuration.yaml in the same folder like this:

```
external_converters:
  - ts0201.js
  ```