pvvx/ZigbeeTLc #17: LYWSD03MMC -sensor freezes ver. 0.1.1.2.

drwalr, 2023-11-26:
I reprogrammed one of my LYWSD03MMC (sensor) to version ver. 0.1.1.2.
After a few hours, the sensor stopped reporting to Z2M.
The connection to the zigbee network i good.

[image]

I changed the zigbee network channel.
Same problem in version 1.08, we restored BLE in all sensor , testing one sensor.

pvvx, 2023-11-26:
> The connection to the zigbee network i good.

Your network contains only routers with Chinese non-standard firmware for Zigbee 3.

drwalr, 2023-11-27:
I have no problems with my network.

"custom routers from China"... I think 99% of devices are made in China

I tested another one firmware.

https://github.com/devbis/z03mmc

I I think this may be a common problem.

https://github.com/devbis/z03mmc/issues/71

pvvx, 2023-11-28:
After temporarily disconnecting the coordinator, the Xiaomi Aqara sensor fell off. It worked through the Tuya smart plug router.
[image]
Where should I go to fix the problem?

drwalr, 2023-11-29:
I'm not a developer, I can't help with the code
Which means
.longPollInterval = 10*4

Are we sure that 10*4 = 10 seconds?
Have you seen the article

https://community.silabs.com/s/article/the-poll-control-cluster-a-reliable-way-for-sed-to-receive-asynchronous-transmi?language=en_US

Something about 7.5 seconds is described?
I use the sonof P coordinator, the latest firmware says something about 7.5 seconds.

[image]

pvvx, 2023-11-29:
> Something about 7.5 seconds is described?
> I use the sonof P coordinator, the latest firmware says something about 7.5 seconds.

Have you seen the article
https://community.silabs.com/s/article/how-often-should-my-end-device-poll-its-parent-x?language=en_US

drwalr, 2023-11-29:
I'll check one more thing.
The sensor has connected to one of the devices, I only have one such model, maybe it is not compliant with the standard, I will turn off the device and pair the sensor again, I will check if the same problem still exists.

[image]

I also have a question, as I wrote, the sensor is frozen and still shows the same temperature and humidity even when I take it in my hand and heat it. If it is not connected to the Zigbee network, the "BT" icon should be displayed as described.?

[image]

Maybe the problem is not with "PollInterval" but the sensor hangs for some other reason. Is there any other way to check this?

pvvx, 2023-11-29:
In the version from this repository, the BLE icon on the screen indicates there is no connection.
The temperature and humidity values on the screen always change with a period of 10 seconds.

drwalr, 2023-12-25:
After updating to 0.1.1.7, everything seems to be OK.

closed 2023-12-25 by drwalr