pvvx/ZigbeeTLc #19: Zigbee not reconize the devices

Benoulimou, 2023-11-28:
Hello, 

I have a problem with the flash of a LYWSD03MMC in zigbee. 

[image]

Z2M doesn't recognize the device and I can't get back to bluethoot. 

Would you have an idea?

pvvx, 2023-11-28:
[README.md](https://github.com/pvvx/ZigbeeTLc#readme):
* To restore to BLE, use the Zigbee OTA file with the same number device from: https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

z2m local OTA:
https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

Benoulimou, 2023-11-28:
thx  ! 
where i find the number device ? 
 sorry

pvvx, 2023-11-28:
[README.md](https://github.com/pvvx/ZigbeeTLc#readme):
| Image Type |  |
| -- | -- |
| 0x020A | LYWSD03MMC ver https://github.com/pvvx/ZigbeeTLc
| 0x0203 | LYWSD03MMC ver https://github.com/devbis/z03mmc

Benoulimou, 2023-11-28:
I'm sorry, I don't understand what to do. I follow what you write but nothing happens. I'm sorry to be a drag.

breaver, 2023-11-28:
As you have 0x020A, you simply download the raw file 1141-020a-99993001-ATC_v46.zigbee 
see: 
https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

You would need to upload it to your Z2M configuration folder and place it there. 
Then add a local ota index file on the same folder, reference it in the configuration.yaml.
In the local index file you need to add:
`[
    {
      "url": "1141-020a-99993001-ATC_v46.zigbee",
      "force": true
    }
  ]`
Restart your Z2M and perform the OTA Update in Z2M for the relevant device.

Benoulimou, 2023-11-29:
> The procedure is more or less described here https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

Thanks for your help. I've read what you've given me but there's only one place where I get stuck. The rest I understood well. It's this one:
 
> In the local index file you need to add: > `[ { "url": "1141-020a-99993001-ATC_v46.zigbee", "force": true } ]`

I can't figure out where I need to add this.

Benoulimou, 2023-11-29:
Good,

I spent my morning looking and I have not found. On the other hand under Home Assistant I installed Z2M normal and not the Edge (for the dev) and the module is very well recognized!

THANK YOU VERY MUCH !!!