pvvx/ZigbeeTLc #27: Bricked CGDK2 / Qingping Temp RH Lite

SkyRaax, 2023-12-08:
I now tried 2x Qingping Temp RH Lite to flash them with the ZigBee Firmware and both sensors just show this screen after flashing:
[image]

I don't have any chance to get this thing working properly. So restarting or resetting don't work.
Also I don't get it in the Zigbee network.

Also the thing don't create any Bluetooth so I can't recover it with the flasher site...
Is there I did wrong or did the manufacturer change something?

pvvx, 2023-12-09:
[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

```
02:51:58: Searching for devices
02:52:35: Connecting to: CGD_136F2B
02:52:39: Hardware Revision String: 2.1.0
02:52:39: Software Revision String: V4.6
02:52:39: Firmware Revision String: github.com/pvvx
02:52:39: Detected custom Firmware
02:52:39: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
02:52:39: Custom config HEX string: 55011000005003a931318678
02:53:09: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0114.bin'...
02:53:10: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0114.bin
02:53:10: File size: 128756 bytes
02:53:10: Count: 8048
02:53:21: Start DFU
02:54:04: Update done after 42.156 seconds
02:54:08: Disconnected.
```
[image]
[image]

[image]
[image]

pvvx, 2023-12-09:
**Back to BLE**

1. configuration.yaml
```
zha:
  zigpy_config:
    ota:
      otau_directory: /home/pvvx/zigpy_ota
```

2. Copy https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-0206-99993001-CGDK2_v46.zigbee to /home/pvvx/zigpy_ota/
3. Restart Home Assistant.
4. 
[image]

```
...
2023-12-09 03:36:37.978 DEBUG (MainThread) [zigpy.zcl] [0x088A:1:0x0019] OTA image_block handler for 'Qingping CGDK2': field_control=2, manufacturer_id=4417, image_type=518, file_version=2576953345, file_offset=58400, max_data_size=48, request_node_addr=None, block_request_delay=0
...
```

5. https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

```
03:39:32: Searching for devices
03:39:53: Connecting to: CGD_136F2B
03:39:57: Hardware Revision String: 2.1.0
03:39:57: Software Revision String: V4.6
03:39:57: Firmware Revision String: github.com/pvvx
03:39:57: Detected custom Firmware
03:39:57: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
03:39:57: Custom config HEX string: 55011000002804a9313186b4
03:40:57: New custom config: [1, 0, 0, 0, 80, 4, 169, 49, 49, 134, 90]
03:40:57: Settings 55010000005004A93131865A was send successful
03:40:58: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
03:40:58: Custom config HEX string: 55010000005004a93131865a00
03:41:01: Send new DevTime ok
03:41:02: Device Date: 2023-12-09 03:41:01.000
03:41:08: Disconnected.
```

KingJ, 2023-12-09:
I am also experiencing the same issue. After flashing with firmware 4.5, the sensor displays a temperature of -3.9c and 10.0% humidity. The flashing log was as follows;

```
16:26:01: Searching for devices
16:26:16: Connecting to: Qingping Temp RH Lite
16:26:53: Searching for devices
16:26:55: Connecting to: Qingping Temp RH Lite
16:26:58: Hardware Revision String: 0001
16:26:58: Firmware Revision String: 2.2.3
16:26:58: Detected Qingping CGDK2 device
16:26:58: Connected
16:27:08: Activating now, please wait...
16:27:12: Registration successful
16:27:12: Send Login, please wait...
16:27:12: Received device infos are correct
16:27:12: Login successful
16:27:21: Load firmware file 'CGDK2_v45.bin'...
16:27:22: File: CGDK2_v45.bin
16:27:22: File size: 81828 bytes
16:27:22: Count: 5115
16:28:35: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0114.bin'...
16:28:35: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0114.bin
16:28:35: File size: 128756 bytes
16:28:35: Count: 8048
16:28:50: Start DFU
16:31:55: Update done after 183.95 seconds
16:32:05: Disconnected.
```

After a few attempts, I did get it to pair with Zigbee2MQTT. However, as the device is unsupported I [took the definition from here for the LYWSD093MMC ](https://smarthomescene.com/guides/convert-xiaomi-lywsd03mmc-from-bluetooth-to-zigbee/)and adapted it to the CGDK2;

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['CGDK2'],
    model: 'CGDK2',
    vendor: 'Qingping',
    description: 'Temp & RH Monitor Lite',
    fromZigbee: [fz.temperature, fz.humidity, fz.battery],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
        await reporting.temperature(endpoint, {min: 10, max: 300, change: 10});
        await reporting.humidity(endpoint, {min: 10, max: 300, change: 50});
        await reporting.batteryVoltage(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.temperature(), e.humidity(), e.battery()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Zigbee2MQTT is now able to read it, but the values are also wrong;

[image]

I wanted to try @pvvx 's instructions [in the previous comment](https://github.com/pvvx/ZigbeeTLc/issues/27#issuecomment-1848015840) to flash it back to BLE but the OTA for this is not yet in Zigbee2MQTT. Therefore, [following the instructions](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files) here I created a custom OTA firmware index with the following contents;

```
[
  {
    "url": "ota/1141-0206-99993001-CGDK2_v46.zigbee"
  }
]
```

I downloaded the firmware file and placed it in to an `ota` folder within the `zigbee2mqtt` folder. You might need to change the URL if you place it elsewhere. I did try setting the URL to the public GitHub one but this fails as Zigbee2MQTT can only extract the firmware info if it is stored locally.

After restarting Zigbee2MQTT I went to the OTA screen and checked for updates for the device. It found one, and I then started the flash process. It takes about 15 minutes. However, once complete the display showed the correct temperature and humidity again, and I was able to reconnect to it over Bluetooth using the [flashing tool](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html);

```
17:30:37: Connecting to: Qingping Temp RH Lite
17:30:44: Hardware Revision String: 2.1.0
17:30:44: Software Revision String: V4.6
17:30:44: Firmware Revision String: github.com/pvvx
17:30:44: Detected custom Firmware
17:30:44: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
17:30:44: Custom config HEX string: 55111000002804a9313186b4
```

@SkyRaax If you are able to get it paired to your Zigbee system, then hopefully you'll be able to get it back on the BLE custom firmware again. It's a little bit more complex for Zigbee2MQTT compared to @pvvx's ZHA instructions but I figured it out eventually! If you're having trouble getting it paired, put it in to pairing mode and place it close to your coordinator - although the screen might go back to the -3.9c display it does eventually seem to pair.

After getting it back on the BLE firmware, I did then try flashing back to the Zigbee firmware, in case I wasn't supposed to have flashed the Zigbee firmware without first flashing the custom BLE firmware. Unfortunately, this produced the same results as before - the sensor displays a temperature of -3.9c and 10.0% humidity. I'm going to try and get it connected back to Zigbee2MQTT again so I can flash the BLE firmware again.

@pvvx Is there anything I can do to help diagnose why the Zigbee firmware doesn't work?

SkyRaax, 2023-12-09:
Yeah that's exactly what happened to me, but thank you for the tip that I can still connect the thing to ZigBee.

Will try this! 

And as it seems I think maybe the manufacturer changed something so that this firmware doesn't work anymore.

pvvx, 2023-12-09:
This means that in these СGDK2 models a different temperature and humidity sensor is installed.

KingJ, 2023-12-09:
I got it back on to the BLE firmware (v4.6) again and sent those commands;

```
18:55:14: Connecting to: Qingping Temp RH Lite
18:55:23: Hardware Revision String: 2.1.0
18:55:23: Software Revision String: V4.6
18:55:23: Firmware Revision String: github.com/pvvx
18:55:23: Detected custom Firmware
18:55:23: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
18:55:23: Custom config HEX string: 55111000002804a9313186b4
18:55:37: Settings 02 was send successful
18:55:37: Sensor: I2C addres 0x70, LCD driver: I2C addres 0x3E
18:55:49: Settings 05 was send successful
18:55:50: Sensor ID: 00000000
```

pvvx, 2023-12-09:
Again Qingping installs poor/substandard sensors...
I'll try to correct it...

pvvx, 2023-12-10:
The source code has been modified ( v 0.1.1.5 + ). Requires CGDK2 test with non-standard sensor.

kiselio, 2023-12-13:
Hello pvvx,

I managed to brick the CGDK2 after trying OTA several times, apparently my model also uses a substandard sensor.
Now I would like to reflash the original or custom BLE firmware, so I can try the Zigbee firmware again.

Can you tell me how to wire the CGDK2 properly? I'm having trouble understanding which wire goes where and which flasher to use.
I have a this adapter at hand: 
[image]

Will this work? Can you give/write a simple guide?
I can solder, I just have no clue where to solder and which flasher to use.

Thank your for your help.
I am happy to provide feedback for this sensor if successful.

pvvx, 2023-12-13:
> Now I would like to reflash the original or custom BLE firmware, so I can try the Zigbee firmware again.

This operation does not require programming via wires.
Zigbee and BLE OTA work on any firmware.

kiselio, 2023-12-13:
I understand, but the device is unable to pair to 2 different coordinators and networks when the Zigbee firmware is flashed.
I assume since the OTA process was interrupted, it got corrupted somehow. Now I am unable to access the device to update via OTA.
For the record, I used a brand new battery

So according to your links, i should attach:

**CGDK2 ---- USB Adapter**
TP17     >>> TX
+Vbat   >>> +3.3V
- Vbat   >>> GND

Which I've already done, But I cannot seem to activate the device for flashing (does not pass activation phase)
Which PIN is the RST on the CGDK2? Do I need to attach that one to the adapter?

Thanks

kiselio, 2023-12-13:
[image]

The process finishes, yet the CGDK2 remains in a bricked state: 

[image]

pvvx, 2023-12-13:
Your adapter does not have an RTC or DCR output.

Therefore, it is necessary to connect TX and +3.3V simultaneously during “activation”.
Or quickly, for a short time, short-circuit RST on the TLSR825x chip while “activation” is in progress.

kiselio, 2023-12-13:
TX terminal on the adapter or TX (TP1) terminal on the CGDK2?
I don't know which terminal is RST

pvvx, 2023-12-13:
http://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/
Download [TLSR8253 Datasheet](http://wiki.telink-semi.cn/doc/ds/DS_TLSR8253-E_Datasheet%20for%20Telink%20BLE+IEEE802.15.4%20Multi-Standard%20Wireless%20SoC%20TLSR8253.pdf)

kiselio, 2023-12-13:
Okay, I erased flash successfully, unplugged and plugged again. At this point, there was nothing on the screen but I was able to connect to COM port. I flashed the smallest firmware successfully (appears so) but then again it shows -3.8C and 10% humidity.
I am doing something very stupid, I know.

pvvx, 2023-12-13:
What firmware did You flash?

kiselio, 2023-12-13:
[CGDK2_v46.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGDK2_v46.bin)

pvvx, 2023-12-13:
This means that this CGDK2 uses some new sensor?

kiselio, 2023-12-13:
I think I'm either failing miserably at flashing (missing something obvious) or this adapter is total trash.
I will get another one from a friend and let you know what I find out.

Thank you very much for your help so far Victor

pvvx, 2023-12-13:
[I flashed CGDK2_v46.bin](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) into a bare TLSR8253:

[image]

https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

Vbat: 3324 mV , Temp: 0.00°C, Humi: 0.00%, ID: 0, Reed switch counter: 0, flg: 0x00:r0/t0

```
15:11:41: Searching for devices
15:11:48: Connecting to: CGD_C1A4F5
15:11:53: Hardware Revision String: 2.1.0
15:11:53: Software Revision String: V4.6
15:11:53: Firmware Revision String: github.com/pvvx
15:11:53: Detected custom Firmware
15:11:53: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHT4x
15:11:53: Custom config HEX string: 55851000002804a9313106b4
15:12:17: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0116.bin'...
15:12:17: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0116.bin
15:12:17: File size: 141924 bytes
15:12:17: Count: 8871
15:12:20: Start DFU
15:12:23: Clear Ext.OTA region (138kib)...
15:12:30: Clean Flash OTA region: 0x040000, size: 140 kbytes. Go OTA...
15:13:15: Update done after 44.812 seconds
15:13:19: Disconnected.
```

kiselio, 2023-12-13:
[image]

I managed to flash it using ATCtelink.py and still getting the -3.8C and 10% hum.

pvvx, 2023-12-13:
You can also flash ATC_v46.bin. BLE will still work.

SkyRaax, 2023-12-15:
BIG shoutout to @KingJ!

Only because of your pretty well documented process of recovering the device I was able to get it back to BLE myself.
So it never was really bricked, thank you!

Another thing:
> The source code has been modified ( v 0.1.1.5 + ). Requires CGDK2 test with non-standard sensor.

@pvvx I tried to flash it again with the flasher itself and I get the same "error" so somehow the sensor isn't still right.
Or maybe I did something wrong :D Here are the logs:

```
11:11:27 AM: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
11:11:27 AM: Custom config HEX string: 55011000002804a9313186b400
11:11:43 AM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0117.bin'...
11:11:43 AM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZCGDK2_v0117.bin
11:11:43 AM: File size: 144644 bytes
11:11:43 AM: Count: 9041
11:12:13 AM: Start DFU
11:12:14 AM: Clear Ext.OTA region (141kib)...
11:12:23 AM: Clean Flash OTA region: 0x040000, size: 144 kbytes. Go OTA...
11:13:59 AM: Update done after 95.359 seconds
11:14:03 AM: Disconnected.
```

In the log you can see I used "ZCGDK2_v0117.bin" this should be the updated version right?

So maybe this information helps you, if you need more, let me know!

SkyRaax, 2023-12-15:
For confirmation here are the output of the Commands 02 and 05:

```
1:24:32 PM: Searching for devices
1:24:42 PM: Connecting to: CGD_803A13
1:24:45 PM: Hardware Revision String: 2.1.0
1:24:45 PM: Software Revision String: V4.6
1:24:45 PM: Firmware Revision String: github.com/pvvx
1:24:45 PM: Detected custom Firmware
1:24:45 PM: Hardware Version: CGDK2 2.1.0, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
1:24:45 PM: Custom config HEX string: 55011000002804a9313186b4
1:25:18 PM: Settings 02 was send successful
1:25:18 PM: Sensor: I2C addres 0x70, LCD driver: I2C addres 0x3E
1:25:24 PM: Settings 05 was send successful
1:25:25 PM: Sensor ID: 00000000
```

pvvx, 2023-12-15:
I couldn't find any of these CGDK2s for sale. In particular, I was not able to find Xiaomi LYWSD03MMC versions B1.5 and B1.6.
The total number of various thermometers purchased for testing, described in the list of supported ones, has long exceeded one hundred. And I'm tired of constantly looking for new store options... 
And it has long been impossible to test all firmware options alone.
I'm waiting for someone who has such a CGDK2 to clarify how to work with its sensor.

joe-sydney, 2023-12-22:
I had the same issue with the -3.9C/10% result. I am not sure if it will help, but this is the same device that had issues with the ATC_MiThermometer firmware where the sensor was SHTV3 (but was eventually fixed by pvvx).

See: https://github.com/pvvx/ATC_MiThermometer/issues/273

pvvx, 2023-12-23:
In the Zigbee version, the previous method does not work.

TheZoker, 2023-12-23:
I have the same issue. I got mine from here, if that helps:
https://aliexpress.com/item/1005005900568131.html

I tried to return to the BLE firmware, but this does not seem to work. I added the firmware file to the OTA folder and installed the ZHA toolkit. 

<details>
  <summary>On the first CGDK2 where I tried this, I got a positive return message:</summary>

```yml
zha_toolkit_version: v1.1.8
zigpy_version: 0.60.1
zigpy_rf_version: 0.22.2
ieee_org: sensor.qingping_cgdk2_battery_2
ieee: 58:f6:3d:81:34:2d:58:5a
command: ota_notify
command_data: null
start_time: "2023-12-23T09:34:45.157909+00:00"
errors: []
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
PAR:
  cmd_id: null
  endpoint_id: null
  dst_endpoint_id: null
  cluster_id: null
  attr_id: null
  attr_type: null
  attr_val: null
  code: null
  min_interval: null
  max_interval: null
  reportable_change: null
  dir: 0
  manf: null
  tries: 1
  expect_reply: true
  args: []
  state_id: null
  state_attr: null
  state_value_template: null
  force_update: false
  allow_create: false
  event_success: null
  event_fail: null
  event_done: null
  fail_exception: false
  read_before_write: true
  read_after_write: true
  write_if_equal: false
  csvfile: null
  csvlabel: null
  download: null
  path: null
  use_cache: false
result: null
success: true
```
</details>

But afterwards I can't find the device within the bluetooth list of the flasher. Do I need to press the button in any way?

<details>
  <summary>On the second one I get this error</summary>

```yaml
zha_toolkit_version: v1.1.8
zigpy_version: 0.60.1
zigpy_rf_version: 0.22.2
ieee_org: sensor.qingping_cgdk2_battery
ieee: 58:aa:39:81:34:2d:58:5a
command: ota_notify
command_data: null
start_time: "2023-12-23T09:40:56.964415+00:00"
errors:
  - >-
    DeliveryError('Failed to deliver packet:
    <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
PAR:
  cmd_id: null
  endpoint_id: null
  dst_endpoint_id: null
  cluster_id: null
  attr_id: null
  attr_type: null
  attr_val: null
  code: null
  min_interval: null
  max_interval: null
  reportable_change: null
  dir: 0
  manf: null
  tries: 1
  expect_reply: true
  args: []
  state_id: null
  state_attr: null
  state_value_template: null
  force_update: false
  allow_create: false
  event_success: null
  event_fail: null
  event_done: null
  fail_exception: false
  read_before_write: true
  read_after_write: true
  write_if_equal: false
  csvfile: null
  csvlabel: null
  download: null
  path: null
  use_cache: false
success: false
```
</details>

Is the only way to fix this by manually flashing them again? Or is there anything I need to do differently?

pvvx, 2023-12-23:
You must proceed as described in "[OTA update in ZHA](https://github.com/pvvx/ZigbeeTLc/issues/7)"
After adding the OTA file to the folder, you need to restart Home Assistant.
After issuing the command, you must wait for the update to take place. The update time varies among different Zigbee networks.

TheZoker, 2023-12-25:
Thanks for the hints!

What solved my problem was changing the path from 
```diff
zha:
  zigpy_config:
    ota:
-      otau_directory: /homeassistant/pvvx/zigpy_ota
+      otau_directory: ./pvvx/zigpy_ota
```

I'm running HA OS and this seems to be the issue. 

Now both my devices are back to the BT firmware. Anything I can do to help you debug the issue with this new sensor?

closed 2025-11-07 by pvvx