pvvx/ZigbeeTLc #28: ZHA: Switch temperature between Celsius and Fahrenheit, Show smiley

pvvx, 2023-12-09:
### Switch temperature between Celsius and Fahrenheit

[image]

Value: 0 -  temperature  in Celsius 
Value: 1 -  temperature  in Fahrenheit

---

### Show smiley

[image]

Value: 0 - Show smiley
Value: 1 - Don't show smiley

(In firmware version from 0.1.1.5)

pvvx, 2023-12-14:
(from firmware version 0.1.1.7)

## Comfort parameters

ClusterID: 0x0204 (Thermostat User Interface Configuration)
Attr: 0x0102, INT8, Temperature minimum, in 1° steps, range -127..+127.
Attr: 0x0103, INT8, Temperature maximum, in 1° steps, range -127..+127.
Attr: 0x0104, UINT8, Hymidity minimum, in 1% steps, range 0..100.
Attr: 0x0105, UINT8, Hymidity maximum, in 1% steps, range 0..100.

[image]

DanTimu, 2024-01-17:
Hello, I'm trying to build the code with a small edit in board_lyws03mmc.h where e set #define SHOW_SMILEY 0 but it is still displayed on the LCD. Is there any possibility not to have the SMILEY/COMFORT displayed on the LCD?

pvvx, 2024-01-17:
ZigBee Cluster Library Specification

Cluster 0x0204 Thermostat User Interface Configuration, Attr: 2 ScheduleProgrammingVisibility
Value: 0 - Show smiley
Value: 1 - Don't show smiley

DanTimu, 2024-01-18:
@pvvx, here are the pictures with #define SHOW_SMILEY 0 and #define SHOW_SMILEY 1
[image]
[image]

broekhuijsen, 2024-03-05:
> @DanTimu
> 
> ZigBee Cluster Library Specification
> 
> Cluster 0x0204 Thermostat User Interface Configuration, Attr: 2 ScheduleProgrammingVisibility Value: 0 - Show smiley Value: 1 - Don't show smiley

That did not work for me, but I managed to change it by changeing this value:

ScheduleProgrammingVisibility.Enabled

to

ScheduleProgrammingVisibility.Disabled

[image]

The manaufacturer code override field does not seem to do anything.

charstring1, 2024-07-22:
Same, manufacturer code override didn't work for me, but set the values from C to F on my LYWSD03MMC using TemperatureDisplayMode.Imperial worked great.