pvvx/ZigbeeTLc #30: ZHA - Setting offsets for temperature and humidity

pvvx, 2023-12-10:
Available starting from firmware version 0.1.1.6.

## Offset for temperature.

The temperature offset is set in 0.1 C steps.
The minimum value is -127 (-12.7 C).
The maximum value is 127 (+12.7 C).

[image]

---

## Offset for humidity.

The humidity offset is set in 0.1 % steps.
The minimum value is -127 (-12.7 %).
The maximum value is 127 (+12.7 %).

[image]

thkrmr, 2023-12-22:
@pvvx sw_build_id = `3001-0117`

ZHA Write Attribute throws an error, logs say `Exception: InCluster 0x0204 not found for '...', endpoint 1`
What could be the issue, do you know? Thank you.

pvvx, 2023-12-22:
If a version update via OTA was used, then the thermometer must be registered again.

The "UserInterface" cluster should appear in "Manage Zigbee device":

[image]

[image]

thkrmr, 2023-12-22:
> If a version update via OTA was used, then the thermometer must be registered again.

@pvvx indeed I pushed the update via OTA. Thank you, will try.

gacopl, 2024-01-11:
hi in Zigbee2MQTT  I get cluster 516 has no command 256, when trying to set offset
i flashed 03MMC with Telink over bluetooth, re registered couple times I am sure i have correct firmware 1118

"softwareBuildID": "3001-0118",

Am i doing something wrong ?

gacopl, 2024-01-26:
> @gacopl  Can you try this PR that add temp/humidifty calibration and others attributes to UI?
> https://github.com/Koenkk/zigbee-herdsman-converters/pull/6939

Don't know how to test tbh using official releases only

naimadfpv, 2024-02-12:
> @gacopl You need to paste code from PR description to new file and add refernece to this file in `customization.yaml`. For details, see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Would be very nice to set these values in the UI. 
Everytime I forgot the value "516" and have to look at it here. 

I also think that a lot of users dont know that this is possible with zha.

naimadfpv, 2024-02-18:
Using pvvx firmware because i switched from zha to z2m. 
If i now calibrate temp offset it’s showing in z2m but not on the display.

f-sh1372, 2024-05-15:
Hi
Thanks for your projects. it is so useull.
I flashed the Zigbee fw using your method. everything works fine but one of my sensors, shows humidity incorrect.
I imported sensor in ZHA using your method.
I have both "skyconnect" and "sonoff zigbee 3.0 USB dongle".
I used zha toolkit and after "Read/Write  Attribute" it shows this log and has error. does it relevant to device's sleep mode?
I changed Tries to 100 but not success. 
I am grateful for your support and guidanc.
[image]

pvvx, 2024-05-15:
ZHA has been updated with Zigpy features.
ZHA Toolkit has not been updated for the new version of Zigpy -> Some ZHA Toolkit functions do not work.

Home Assistant Core logs:
```
WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration zha_toolkit which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```

pvvx, 2024-05-15:
After updating via Zigbee OTA to version 0.1.2.1, it is advisable to delete the device in the ZHA and re-pair the device.

[image]

```
zha_toolkit_version: v1.1.10
zigpy_version: 0.63.5
zigpy_rf_version: 0.38.1
ieee_org: update.xiaomi_lywsd03mmc_z_firmware
ieee: a4:c1:38:0b:5e:ed:85:b7
command: attr_read
command_data: null
start_time: "2024-05-15T07:52:59.020252+00:00"
errors: []
params:
  endpoint_id: 1
  cluster_id: 516
  attr_id: 257
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
attr_type: "0x29"
write_is_equal: false
result_read:
  - "257": 0
  - {}
success: true
```

Meisam-kh, 2024-05-16:
Thanks for your support. I  deleted the device in the ZHA and re-pair the device. After that, Calibration was successful.
but there is a problem. is there any solution to decrease time of update interval for Home assistant reading sensor data?
It upadate sensor datas after 10 to 20 minutes (of course after data change). I would replace the battery with an adapter and I don't care about power consumption.
in the below picture it is shows that last update is about 9 min and 21 min
[image]

pvvx, 2024-05-16:
1. Set the measurement interval to 3 seconds.
2. Set the minInterval of the report to 0.

(!) In some cases, when the data is rounded, the update in HA does not occur for the same values.

pvvx, 2024-05-17:
> 1- can I just set the measurement interval & min Interval without doing the patch?
> 3- for setting "min interval of the report" . would you please help me more?
(show me which fields and how they should fill?)

When connecting a device or pressing the “RECONFIGURE” button, the ZHA sets the reporting intervals and delta changes specified in the internal code.
Single adjustment of measure intervals will not help.

---

For short intervals of temperature and humidity reporting, the best solution is the BLE option. The BLE firmware by default transmits new measurements every 10 seconds. You can reduce it to 5 seconds without changing the battery to a more powerful one. Using an external power source, you can install measurement and transmission intervals for 0.5 seconds.

Danstiv, 2024-10-18:
Hello @pvvx!
I see cluster 516 in ZHA UI, but it only has 3 attributes:

1. keypad_lockout (id: 0x0001).
2. schedule_programming_visibility (id: 0x0002).
3. temperature_display_mode (id: 0x0000).￼￼

Is this some kind of configuration issue? Or do I need to install ZHA Toolkit?

Hardware:
- SONOFF_Zigbee_3.0_USB_Dongle.
- LYWSD03MMC (B 1.5)

So how to change offsets and measurement frequency, now, at the end of 2024, can you give instructions?
Thanks.

pvvx, 2024-10-18:
> Is this some kind of configuration issue? Or do I need to install ZHA Toolkit?

install ZHA Toolkit.