pvvx/ZigbeeTLc #31: Flashing LYWSD03MMC with zigbee firmware using Telink Flasher v7.3 gives error: "Size firmware is more 131072 bytes!"

matrover, 2023-12-12:
Hi, 

The title says it all, i've done this flashing a few times but this time it gives me the error: Size firmware is more 131072 bytes!

Any idea? 

Thanks!

pvvx, 2023-12-13:
https://github.com/pvvx/ZigbeeTLc#to-flash-the-zigbee-firmware-use-a-google-chrome-microsoft-edge-or-opera-browser

_If the firmware file is larger than 128 kilobytes, then you will have to download BLE version 4.6 or later. BLE versions below 4.6 do not support OTA downloads with fw sizes larger than 128 kilobytes. Xiaomi firmwares, excluding MJWSD05MMC, also do not support OTA downloads with fw sizes larger than 128 kilobytes._

matrover, 2023-12-13:
Great will try, thanks!

matrover, 2023-12-15:
Hi i updated the sensor, but it now says: 

`14:37:50: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 5, CtrID: 0x5858 Not bound, Standard certification, MAC: A4C138BFE948, Data encrypted
14:38:02: Hardware Revision String: B1.4
14:38:02: Software Revision String: 0159
14:38:02: Firmware Revision String: 2.1.1_0159
14:38:02: Detected Mi device
14:38:02: (!) Version: 2.1.1_0159 is temporarily not supported (!)`

lewicki-pk, 2023-12-23:
> Hi i updated the sensor, but it now says:
> > 14:38:02: Firmware Revision String: 2.1.1_0159
> > 14:38:02: Detected Mi device
> > 14:38:02: (!) Version: 2.1.1_0159 is temporarily not supported (!)

this version unfortunately only allows wired upload of firmware (guess- I didn’t test it). OTA updates don’t work with this one. I think someone needs to find a version that is in the middle or upload BLE version first with TelinkMiFlasher - I’ll test the latter in a minute 😀

lewicki-pk, 2023-12-23:
BLE custom software installed using TelinkMiFlasher does not allow further upload of Zigbee firmware. Maybe this could be a feature of newer version ?

pvvx, 2023-12-23:
> BLE custom software installed using TelinkMiFlasher does not allow further upload of Zigbee firmware. Maybe this could be a feature of newer version ?

https://github.com/pvvx/ATC_MiThermometer
 * _Firmware download support for [Zigbee](https://github.com/pvvx/ZigbeeTLc)_

lewicki-pk, 2023-12-24:
@pvvx I think we don’t understand each other here. 

Let me explain my case (and maybe @matrover has the same case):
With the new device I’m not able to install Zigbee firmware because of the size of the OTA update. To be able to have bigger uploads I need a newer version of original firmware. But (!) updating to latest version 2.1.1_0159 does not allow OTA updates. 
So I thought that maybe I will update the firmware to a different option available in TelinkMiFlasher- BLE. But that doesn’t work because from BLE software I cannot make an update to Zigbee. 

Last option is the third one possible via TelinkMiFlasher- some Xiaomi software. Maybe then it will work but the sensor where I uploaded the BLE firmware is “locked” until I either upload firmware with UART or BLE software gets an update that allows OTA flashing of Zigbee firmware.

pvvx, 2023-12-24:
You don't need new original firmware from Xiaomi. You need alternative firmware version 4.6, which is provided by TelinkMiFlasher.

lewicki-pk, 2023-12-24:
Thanks @pvvx for that info. I’ll try today with 4.6.

lewicki-pk, 2023-12-26:
FYI: you were right - I haven't seen the option to update to v4.6 and later on to Zigbee but I've tested it yesterday and it works flawlessly!
Thanks