pvvx/ZigbeeTLc #33: Unable to upgrade from devbis to ZigbeeTLc with OTA / ZHA

CLARENNE-Q, 2023-12-12:
Hello,

I am experiencing difficulties when using OTA (Over-The-Air) to flash the firmware on LYWSD03MMC.

For beginning, i flashed the device with devbis/z03mmc but sensor crash and don't show true temperature/humidity.

I want to use pvvx/ZigbeeTLc because i never experiment issue with the pvvx BLE  version !

[image]

 When attempting the update, I encounter the following issues:

My configuration.yaml:

[image]

The wigpy_ota folder:
[image]

I put zigbee in folder like this:

[image]

The Service ZHA Called:

[image]

The error in Logviewer:

[image]

Observed Behavior:

However, the update process fails at a certain stage, and the LYWSD03MMC is not updated correctly.

pvvx, 2023-12-13:
Devbis version has a different ID.
https://github.com/pvvx/ZigbeeTLc#zigbee-ota

_To restore to BLE, use the Zigbee OTA file with the same number device from:_ https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

1. Recovery to BLE version 4.6 using file:
https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-0203-99993001-ATC_v46.zigbee

2. Use [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) to flash new Zigbee version.

CLARENNE-Q, 2023-12-13:
Thanks for your fast reply, i will try tomorrow !

CLARENNE-Q, 2023-12-23:
I have HomeAssitant OS and all the time i had this error:

`No OTA image available`

This is because the otau_directory for Home Assistant OS must be set like this:

`zha:
  zigpy_config:
    ota:
      otau_directory: ./zigpy_ota`

not like this:

`      otau_directory: /zigpy_ota`

Thanks to https://sonoff.tech/product-review/tutorial/local-configure-zigbee-ota-in-home-assistant-tutorial/

pvvx, 2023-12-23:
> not like this:
      otau_directory: /zigpy_ota

Nobody bothers you to create a folder in the root directory and not look for where Home Assistant is installed.

CLARENNE-Q, 2023-12-26:
I have 2 other LYWSD03MMC with ZigbeeTLc:

[image]

I want to restore to ATC_MiThermometer  because BTE work better with my setup.

I put all zigbee files here:

[image]

But the OTA image not loaded:

`2023-12-26 10:27:40.298 DEBUG (MainThread) [custom_components.zha_toolkit] Got hass.data['zha']/gateway <homeassistant.components.zha.core.gateway.ZHAGateway object at 0x7fde0e201290>
2023-12-26 10:27:40.317 DEBUG (MainThread) [custom_components.zha_toolkit] module is <module 'custom_components.zha_toolkit' from '/config/custom_components/zha_toolkit/__init__.py'>
2023-12-26 10:27:40.321 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Parameters '{'ieee': 'sensor.arthur_zigbeetlc_battery_2'}'
2023-12-26 10:27:40.321 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Final manf 'None'
2023-12-26 10:27:40.322 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Found registry_entity RegistryEntry(entity_id='sensor.arthur_zigbeetlc_battery_2', unique_id='a4:c1:38:45:42:b4:f3:a3-1-1', platform='zha', previous_unique_id=None, aliases=set(), area_id=None, capabilities={'state_class': 'measurement'}, config_entry_id='3bfaf53946370617e74f338226080d03', device_class=None, device_id='af41fa67e549e62ecd424afe61ee2939', disabled_by=None, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, hidden_by=None, icon=None, id='33298dc9a88e83334f9cf6e28f6bcdc9', has_entity_name=True, name=None, options={'cloud.alexa': {'should_expose': False}, 'cloud.google_assistant': {'should_expose': False}, 'conversation': {'should_expose': False}}, original_device_class='battery', original_icon=None, original_name='Battery', supported_features=0, translation_key=None, unit_of_measurement='%')
2023-12-26 10:27:40.322 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Found registry_device DeviceEntry(area_id='chambre_arthur', config_entries={'3bfaf53946370617e74f338226080d03'}, configuration_url=None, connections={('zigbee', 'a4:c1:38:45:42:b4:f3:a3')}, disabled_by=None, entry_type=None, hw_version=None, id='af41fa67e549e62ecd424afe61ee2939', identifiers={('zha', 'a4:c1:38:45:42:b4:f3:a3')}, manufacturer='Xiaomi', model='LYWSD03MMC', name_by_user='Arthur ZigbeeTLc', name='Xiaomi LYWSD03MMC', serial_number=None, suggested_area=None, sw_version='0x01173001', via_device_id=None, is_new=False)
2023-12-26 10:27:40.322 DEBUG (MainThread) [custom_components.zha_toolkit] 'ieee' parameter: 'sensor.arthur_zigbeetlc_battery_2' -> IEEE Addr: 'a4:c1:38:45:42:b4:f3:a3'
2023-12-26 10:27:40.323 DEBUG (MainThread) [custom_components.zha_toolkit] Default handler for ota_notify
2023-12-26 10:27:40.323 DEBUG (MainThread) [custom_components.zha_toolkit] Handler: <function command_handler_default at 0x7fdde1b27c40>
2023-12-26 10:27:40.323 DEBUG (MainThread) [custom_components.zha_toolkit] running default command: <ServiceCall zha_toolkit.ota_notify (c:01HJKBT6R85HZRK7573GBF0W1R): ieee=sensor.arthur_zigbeetlc_battery_2>
2023-12-26 10:27:40.324 DEBUG (MainThread) [custom_components.zha_toolkit.default] Trying to import custom_components.zha_toolkit.ota to call ota_notify
2023-12-26 10:27:40.326 DEBUG (MainThread) [custom_components.zha_toolkit.ota] OTA_notify
2023-12-26 10:27:40.359 DEBUG (MainThread) [custom_components.zha_toolkit.ota] running 'image_notify' command: <ServiceCall zha_toolkit.ota_notify (c:01HJKBT6R85HZRK7573GBF0W1R): ieee=sensor.arthur_zigbeetlc_battery_2>
2023-12-26 10:27:40.360 DEBUG (MainThread) [custom_components.zha_toolkit.utils] IEEE for get_device: a4:c1:38:45:42:b4:f3:a3 a4:c1:38:45:42:b4:f3:a3
2023-12-26 10:27:40.360 DEBUG (MainThread) [custom_components.zha_toolkit.utils] Tries remaining: 1
2023-12-26 10:28:01.421 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] Received ZCL frame: b'\x01\xd8\x01\x00A\x11\n\x02\x010\x17\x01'
2023-12-26 10:28:01.422 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=216, command_id=1, *direction=<Direction.Server_to_Client: 0>)
2023-12-26 10:28:01.422 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] Decoded ZCL frame: Ota:QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4417, image_type=522, current_file_version=18296833)
2023-12-26 10:28:01.423 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] Received command 0x01 (TSN 216): QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4417, image_type=522, current_file_version=18296833)
2023-12-26 10:28:01.423 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=4417, image_type=522, current_file_version=18296833, hardware_version=None, model='LYWSD03MMC'
2023-12-26 10:28:01.425 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] No OTA image is available
2023-12-26 10:28:01.426 DEBUG (MainThread) [zigpy.zcl] [0x8576:1:0x0019] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=216, command_id=2, *direction=<Direction.Client_to_Server: 1>)`

Do you have an idea for this ?

pvvx, 2023-12-26:
After placing the OTA files, it is advisable to restart Home Assistant.
The ZHA documentation and the ZHA toolkit say that this is not necessary, but without this action OTA does not work for me.

fhempy, 2024-01-30:
> Devbis version has a different ID (Image Type). https://github.com/pvvx/ZigbeeTLc#zigbee-ota
> 
> Zigbee integration does not allow the device to be reprogrammed with different IDs (Image Type). And "File Version" should be a large number.
> 
> _To restore to BLE, use the Zigbee OTA file with the same number device from:_ https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota
> 

Hi,
I would like to migrate to your firmware as well, but the local OTA with zigbee2mqtt doesn't work. Can you please let me know which zigbee_ota file I need to use? I couldn't find any version information which would match the files there!?
Thanks!

pvvx, 2024-01-31:
[README](https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#zigbee-ota)
1. To restore to BLE, use the Zigbee OTA file with the same number device from: https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota
LYWSD03MMC -> https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-020a-99993001-ATC_v46.zigbee and https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-0203-99993001-ATC_v46.zigbee
2. Via BLE using [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) you can record any version of Zigbee.

pvvx, 2024-02-13:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis

djbuldog, 2024-02-17:
I also want to upgrade from devbis fw 1.1.0 to ZigbeTLc fw or back to custom BT 4.6 but there is a strange manufacturer_id=56085 in the log which is probably the reason of not finding any OTA image:

```
2024-02-17 11:33:14.782 DEBUG (SyncWorker_17) [zigpy.ota.provider] ImageKey(manufacturer_id=4417, image_type=515): ./zigbee_ota/1141-0203-09993001-Z03MMC_v0120.zigbee, version: 161034241, hw_ver: (None, None), OTA string: Zigbee ver:devis to ZigbeeTLc
2024-02-17 12:12:24.226 DEBUG (SyncWorker_3) [zigpy.ota.provider] ImageKey(manufacturer_id=4417, image_type=515): ./zigbee_ota/1141-0203-99993001-ATC_v46.zigbee, version: 2576953345, hw_ver: (None, None), OTA string: Zigbee ver:devis to BLE
..
2024-02-17 12:12:28.599 DEBUG (MainThread) [zigpy.zcl] [0xEC8B:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=56085, image_type=515, current_file_version=285224961, hardware_version=None, model='LYWSD03MMC'
2024-02-17 12:12:28.600 DEBUG (MainThread) [zigpy.zcl] [0xEC8B:1:0x0019] No OTA image is available
```
current_file_version and image_type is correct... but what is manufacturer_id=56085?
Does anybody know anything about it? Is the value coming from devbis fw or hw?

djbuldog, 2024-02-17:
Let me share my issue (SOLVED) with upgrading - it can help somebody - maybe. I wanted to migrate from devbis fw 1.1.0 to zigbeetlc over OTA/ZHA but no OTA image was found due to different manufacture_id 56085 vs 4417. The solution was select the right image :D with db15 prefix (56085 = 0xDB15).. as described in https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis.. 

```
2024-02-17 11:33:14.782 DEBUG (SyncWorker_17) [zigpy.ota.provider] ImageKey(manufacturer_id=4417, image_type=515): ./zigbee_ota/1141-0203-09993001-Z03MMC_v0120.zigbee, version: 161034241, hw_ver: (None, None), OTA string: Zigbee ver:devis to ZigbeeTLc
2024-02-17 12:12:24.226 DEBUG (SyncWorker_3) [zigpy.ota.provider] ImageKey(manufacturer_id=4417, image_type=515): ./zigbee_ota/1141-0203-99993001-ATC_v46.zigbee, version: 2576953345, hw_ver: (None, None), OTA string: Zigbee ver:devis to BLE
..
2024-02-17 11:33:19.079 DEBUG (MainThread) [zigpy.zcl] [0xEC8B:1:0x0019] OTA query_next_image handler for 'Xiaomi LYWSD03MMC': field_control=0, manufacturer_id=56085, image_type=515, current_file_version=285224961, hardware_version=None, model='LYWSD03MMC'
2024-02-17 11:33:19.079 DEBUG (MainThread) [zigpy.zcl] [0xEC8B:1:0x0019] No OTA image is available
```