pvvx/ZigbeeTLc #35: Config endpoints (either undocumented or not implemented)

sjthespian, 2023-12-14:
I have a test Mijia that I have converted to Zigbee, and I'm finding the lack of configuration painful... I've figured out some of the endpoints I can tweak with the zha-toolkit, but I'm wondering if the others are not yet implemented or just not documented?

Here are the ones I scraped from the docs and from other issues:
```
ClusterID: 0x0204 (Thermostat User Interface Configuration), Attr: 0x0000 (TemperatureDisplayMode), ENUM8 = 0 in C, = 1 in F,
                                                             Attr: 0x0002 (ScheduleProgrammingVisibility), ENUM8 = 0 show smiley On, = 1 show smiley Off)
                                                             Attr: 0x0100, INT8, Temperature offset, in 0.1° steps.
                                                             Attr: 0x0101, INT8, Humidity offset, in 0.1% steps.
```

Looking at the configuration values I had from one of the BT devices, are any of these implemented?
* Smiley (comfort / custom)
* Show clock (and also a way to set the time)
* Show battery
* Screen off
* Advertising type and interval (may not make any sense with Zigbee)
* Measure interval
* Connect latency
* RF TX Power
* Minimum LCD refresh rate
* Comfort parameters:
** Temperature Lo / Hi
** Humidity Lo / Hi

It would also be nice to have a way to control the interval of switching between clock/battery/humidty info -- i don't see a setting for that in the BT version. It's possible it's a side effect of one of the other settings like measure interval though.

And is there a way to put custom data on the device screen? (ex. the BT custom firmware has a "Show on device screen" function where you can put specific values for the temperature, smiley, humidity, and battery).

I don't use all of these, but I'm sure there are folks who do. And I figured I may as well save you having to deal with 20 different feature requests if you can do them all in one... :-)

pvvx, 2023-12-14:
More than half of the above list will not be in the Zigbee version.

pvvx, 2023-12-14:
https://github.com/pvvx/ZigbeeTLc/issues/28#issuecomment-1855763432

closed 2023-12-14 by pvvx