pvvx/ZigbeeTLc #36: Can't pair when flashing 0.1.1.7

schumijo, 2023-12-14:
Hi

I flashed several LYWSD03MMC last days with version 0.1.1.6 with success.
But I just tried to flash a new one with 0.1.1.7 (I didn't notice a new version was available) and I can't pair with Z2M. I tried to remove battery,  and also shortcut GND and reset without success

Any idea ?

pvvx, 2023-12-14:
After 7 seconds, when connecting the “reset” and “gnd” contacts, the thermometer displays “---”?

schumijo, 2023-12-15:
Nothing happened after 7 seconds.
It's not an upgrade but a fresh install

pvvx, 2023-12-15:
LYWSD03MMC hw B1.4..B2.0 ?

https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#action-of-the-button

schumijo, 2023-12-15:
B1.4
I did a lot of tests today without success
Now, I managed to go back to bluetooth and I am trying to flash again but I am stuck with "GATT Error Unknown"
Still investigating, keep you informed

pvvx, 2023-12-15:
I cannot recreate the problems described.
Groups of the thermometers allocated for testing new firmware work without problems.
Here is a group of devices tested together with B1.4 installed on the coldest wall of the house:
[image]
The one with the wire sticking out is B1.4, which has been reprogrammed a thousand times in 3 years....

schumijo, 2023-12-16:
Hi @pvvx

I finally managed to flash this thermometer with 0.1.1.6 then ota update to 0.1.1.7
I really don't understand what happened with this one. I flashed all my 7 thermometers without problem.

Oh really interesting this version with both BLE and Zigbee. Will test it on one of my LYWSD03MMC

Thank you for your hard work !

twielandmedicusan, 2024-01-09:
Hi @pvvx 

I have a similar problem. I flashed 4 [LYWSD03MMC] thermometers last week with the web tool and one today (with the default zigbee version that's available in the tool). The ones from last week joined zigbee2mqtt without any problem and the one today doesn't join. It shows the bluetooth sign (not flashing) and I reset it a couple of times already by shorting gnd and reset and removed the battery for a while, but no change. Anything else I could try? I also don't know how I can re-flash given that bluetooth is not available anymore