pvvx/ZigbeeTLc #37: Battery life

tristone-cz, 2023-12-18:
Hi all.

I had no troubles to upgrade the LYWSD03MMC to custom BLE and then to 1.1.4.
Z2M works fine, even upgrade to 1.1.7 was relatively smooth.

But what about battery life?
The original battery was depleted in 10 and 12 days (two devices).

Do you have the same experience?
Or there is something wrong with my ones?

Both are placed very near the Zigbee stick, LQI is around 150-200.

pvvx, 2023-12-18:
Each Zigbee OTA update requires 20 times more power than BLE OTA.
Several test thermometers with Zigbee have been working for more than a month. There is no critical battery discharge on them.

On Xiaomi LYWSD03MMC B1.4 a month ago it was 2.82V. Today 2.79V. Before this, the battery worked for several months in the BLE version.

tristone-cz, 2023-12-18:
OK. Is there  some way to influence the update frequency? I did not found in documentation.
These thermometers will not be used in any critical env so even slow (each °C) updates would be OK.

pvvx, 2023-12-18:
Compared to the BLE version with default settings (transmission every 2.5 seconds, sensor polling 10 seconds), Zigbee consumption (communication step every 10 seconds, sensor polling 10 seconds) is one and a half times 1.5 times higher.
Plus the dependence in Zigbee on the settings of the minimum and maximum interval and change delta.

Each iteration of transmission and reception in Zigbee requires 4 times more energy than in BLE. And since Zigbee uses a more powerful signal for transmission in dBm, the differences become even greater.

pvvx, 2023-12-18:
I do not have long-term records of CR2032 battery consumption on thermometers with Zigbee firmware.
There are short measurements between different firmware versions.
[For BLE option with Zigbee](https://github.com/pvvx/BZdevice) and battery CR2032:
[image]

tristone-cz, 2023-12-18:
OK, thanks for detailed info.
I replaced batteries and will watch it. 
Weird is that on the news ones is immediately reported as 40%. I will pick up some serious brand for next round 😄

pvvx, 2023-12-18:
Battery charge values are reported using average of 256 measurements. Otherwise the graph will be very noisy.

JayFoxRox, 2023-12-18:
> The original battery was depleted in 10 and 12 days (two devices).

I have been using the bluetooth version for a couple of months already. I don't think I ever swapped the preinstalled battery.

I flashed 9x LYWSD03MMC with the zigbee version last week.
If the zigbee version really dies this quickly then I'll return to the bluetooth version.

Currently the batteries report (in zigbee2mqtt) as 92%, 88.5%, 100%, 86%, 100%, 100%, 98.5%, 88% and 100%.
They all run the zigbee version with the preinstalled battery except the first one (which uses some fresh no-name battery).
They are all B1.7 (I think?) and were manufactured in 2023.02 according to the packaging, again, except the first one (not sure which HW revision).

> When using any of the Zigbee chips, significantly more battery power is required to meet the standards and functionality of routers and coordinators on the market than devices operating in the Bluetooth v5.
> The communication range of Zigbee is 3-4 times less than that of Bluetooth 5.0 in LE Long Range.
> And with the release of Bluetooth v5.4, Zigbee becomes completely unnecessary.

Are you saying we should almost always prefer to operate these devices with bluetooth?
Why put so much effort in the zigbee firmware then?

Personally I'm using a lot of zigbee lamps / switches for cost reasons (although I consider switching buttons to cheap bluetooth camera shutter switches). I also don't want WiFi devices for security/privacy reasons.

I reflashed the zigbee version for a number of reasons:
Primarily, setting up a bluetooth bridge is really annoying and I'm also concerned with the range of bluetooth (as I have many zigbee devices which extend the mesh, but I don't have any powered bluetooth devices).
I knew that zigbee would consume slightly more battery, however, I'm also hoping that battery performance will be optimized over time (hopefully more customization options).

If bluetooth is truly better, then I'd probably return to that.
Both, zigbee and bluetooth end up in MQTT on the same raspberry pi for me anyway.
However, I had problems with sharing my bluetooth device with multiple docker containers in the past, so zigbee using zigbee2mqtt is simpler.

I think it would be great if the README explained the actual benefits. Things like:
- With bluetooth, what is the expected range and when should one use zigbee?
- How long is battery life in practice with the same kind of battery in the bluetooth version vs zigbee?
- ...

> In Zigbee firmware, the sensor is polled every 10 seconds to display and analyze measurements.
> This interval is fixed. As this interval increases, the display will no longer correspond to the measurements.

Personally I don't care too much about the display. I'd rather have a long battery time and the measurements taken every 60 seconds or so.
Temperature doesn't change too quickly where most of my thermometers are installed anyway.

I'd be much more concerned with the accuracy of the displayed values (so I'd like to see 2-point calibration mode so the device displays more accurate values).

I think a hybrid mode would also be cool, where it would display the clock for 30 seconds or so, and then when the next temperature measurement is ready, it would display the temperature again.

> For this reason, Linux has not allowed users to use the new Bluetooth standards since 2016.

I'm not sure what you mean here? Doesn't linux support them? Also if they are so good, why wouldn't they be supported in Linux?

pvvx, 2023-12-18:
> Are you saying we should almost always prefer to operate these devices with bluetooth?
> Why put so much effort in the zigbee firmware then?

I haven't put much effort into the Zigbee firmware.
And I previously described that this would not be a very correct path for thermometers with CR2032. But there was an advertisement about devbis firmware. And I’m just trying to optimize consumption and expand functions.

> Personally I don't care too much about the display. I'd rather have a long battery time and the measurements taken every 60 seconds or so.

This greatly limits the scope of application of these thermometers. 
If the interval transmission of results exceeds 10 seconds, the thermometer becomes a game thermometer and is not applicable in IoT.
In such conditions, there is no point in transmitting measurements at all, because sometimes it is enough to look at the display.

> I'm not sure what you mean here? Doesn't linux support them? Also if they are so good, why wouldn't they be supported in Linux?

I don't look into world conspiracies. :)

pvvx, 2023-12-18:
> I think a hybrid mode would also be cool, where it would display the clock for 30 seconds or so, and then when the next temperature measurement is ready, it would display the temperature again.

Measuring and displaying does not significantly affect battery consumption. And it does not create large pulsed currents, which have a bad effect on the operating life of CR2032 series batteries.

pvvx, 2023-12-18:
> I think it would be great if the README explained the actual benefits. Things like:
> * With bluetooth, what is the expected range and when should one use zigbee?
> * How long is battery life in practice with the same kind of battery in the bluetooth version vs zigbee?

It is impossible to answer these questions unambiguously.
The charge level of the CR2032 battery from different manufacturers and operating conditions vary many times. Some batteries provide only 15% of the capacity of others.

Xiaomi specifically does not install a capacitor in the power circuit so that the use of the battery is limited to 60% of its capacity. At a level of 40% of the actual capacity, the battery is discarded, since with a pulse of transmission current there is no longer enough voltage to operate the chips used. Chip manufacturers and thermometer developers provided space for installing a capacitor, but marketing...

alexw1982, 2023-12-19:
> Measured average consumption on Xiaomi LYWSD03MMC B1.4:
> 
>     1. Firmware version [BZdevice](https://github.com/pvvx/BZdevice) - 18..20 µA.
> 
>     2. Firmware option [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc) - 14..25 µA.
> 
>     3. [BLE firmware](https://github.com/pvvx/ATC_MiThermometer) option - 14 µA.
> 
>     4. Firmware option [z03mmc](https://github.com/devbis/z03mmc) - 20..40 µA.

So your zigbee firmware showed almost the same power consumption during normal operations (not including update / flashing) - is that the correct intepretation?

pvvx, 2023-12-19:
Not really. 14 µA is obtained if there is practically no transmission between the thermometer and the coordinator.
This can be done artificially - do not subscribe to the transfer of measurements.

slingel, 2023-12-19:
I'm seeing around 1mA with all your firmwares after interview with Sonoff CC2652P dongle on all my LYWSD03MMC.
Every few seconds current drops below 10µA then right back to 1mA.
With devbis firmware devices immediately sleep at less than 10µA after interview. 
I haven't tried 1117 yet.
TS0201 is not affected.

pvvx, 2023-12-19:
https://github.com/devbis/z03mmc/issues/11#issuecomment-1804344704

tristone-cz, 2023-12-20:
Just some graph of original and then "new" Chinese battery. Today I will put there something more serious and let's see.
But the same batteries in many other Aqara zigbee devices work for years.
[image]

EddieTH, 2023-12-20:
> Xiaomi specifically does not install a capacitor in the power circuit so that the use of the battery is limited to 60% of its capacity. At a level of 40% of the actual capacity, the battery is [...]

Have you done trials with adding such a capacitor? Would there actually be a significant improvement?

pvvx, 2023-12-20:
> Have you done trials with adding such a capacitor? Would there actually be a significant improvement?

https://www.ti.com/lit/wp/swra349/swra349.pdf

https://github.com/devbis/z03mmc/issues/11#issuecomment-1841546441

pvvx, 2023-12-21:
## Aqara lumi.weather

Sleep - 1.8 µA

When transmitting nothing - Average - 12.5 µA
[image]

When data is being transferred:
[image]

The capacitor is installed in the power circuit.

pvvx, 2023-12-21:
Based on the measurements taken, you can calculate how much LUMI will consume when transmitting measurements every 180 seconds.
This will be (12.5*170+74.8*10)/180 = 15.96 µA
When transmitting every 30 seconds (12.5*20+74.8*10)/30 = 33.27 µA   :P

tristone-cz, 2023-12-21:
@pvvx I am using the LYWSD03MMC, version not exactly sure. I think the FW version during flashing was 1.7 or 1.8

alexw1982, 2023-12-22:
> Based on the measurements taken, you can calculate how much LUMI will consume when transmitting measurements every 180 seconds. This will be (12.5 x 170+74.8 x 10)/180 = 15.96 µA When transmitting every 30 seconds (12.5 x 20+74.8 x 10)/30 = 33.27 µA :P

I have no calculations to show for but I have multiple award zigbee temp/humidity sensors running for >1yr (even in the freezer at -18C)

pvvx, 2023-12-22:
> I have no calculations to show for but I have multiple aqara zigbee temp/humidity sensors running for >1yr (even in the freezer at -18C)

1. A clock that doesn't go at all shows the correct time twice a day.
2. If LUMI starts transmitting measurements every 180 seconds, then the consumption will be the same as that of the BLE version of Xiaomi LYWSD03MMC. At the same time, Xiaomi LYWSD03MMC BLE transmits measurements every 10 seconds. The difference will be 18 times. :P

pvvx, 2023-12-23:
@alexw1982 - Tell me how to control heating using the Aqara sensor?

[image]
[image]

Neo2SHYAlien, 2023-12-23:
I'm not particularly adept at understanding electrical diagrams, but I believe this area is where the capacitor should be placed. (small one). The is 1.4 version LYWSD03MMC. 
[image]

> https://www.ti.com/lit/wp/swra349/swra349.pdf

pvvx, 2023-12-23:
The installation of two capacitors is provided. С24 and С25.
[image]

victorlucachi, 2023-12-27:
> But what about battery life? The original battery was depleted in 10 and 12 days (two devices).

I'm wondering the same thing. I currently have just one sensor that I serial flashed with the latest Zigbee firmware, and the battery is draining rather fast; first one was the included chinesium, the second was a Varta.

[image]

pvvx, 2023-12-27:
What "min/max/change" settings were used?
[image]

victorlucachi, 2023-12-27:
> What "min/max/change" settings were used?

1. everything at defaults, I sent a command only to remove the smiley face from the display, but I had a feeling that might've glitched the sensor somehow (and hence the battery drain) so I removed it from the zigbee network, and repaired it.
2. the coordinator is a sonoff zigbee dongle E
3. probably around 3m, through a rather thick concrete wall

pvvx, 2023-12-27:
> everything at defaults,

What's in reality? Z2M and ZHA set their own settings.

victorlucachi, 2023-12-27:
> What's in reality? Z2M and ZHA set their own settings.

oh, I apologise, it's my first zigbee device and I'm still trying to wrap my head around it.

this is what turns up if I attempt to reconfigure the device in ZHA

[image]

pvvx, 2023-12-27:
What is the hw_version of the thermometer?
B1.4....B2.0?
[image]

victorlucachi, 2023-12-27:
it's a B1.6

pvvx, 2023-12-27:
I don't have B1.5 and B1.6.
At B1.4 and B2.0 the consumption is 14..20 µA.
This is enough to last more than 6 months with the cheapest CR2032.

victorlucachi, 2023-12-27:
fair enough. thanks for the input nevertheless

I'm about to receive 4 more thermometers (maybe before the end of year if I'm lucky), and if they are a different hardware revision I will report back.

6 months of battery life would be really nice to have.

pvvx, 2023-12-27:
1. Try to reduce the distance to the Zigbee coordinator and exclude routers.
2. Try the [firmware from devbis](https://github.com/devbis/z03mmc). There is a difference from B1.6.
3. Try the light version 0.1.1.8. (Options disabled: "Groups", "TouchLink", "Find and Bind".)

victorlucachi, 2023-12-27:
noted. I will probably wait for the other 4 sensors to arrive, and I will flash them with different firmware releases, to compare power usage, after which I'll report back

thank you again, much appreciated!

EddieTH, 2023-12-27:
[image]

I have 4 Xiaomi thermometers LYWSD03MMC hw1.4, sw ver 0.1.1.1. My monthly statistics are very promising according to me.
The best result is an increase (!) in battery status from 84% to 87%, a sensor with a new battery lost only 2%, another one (this one seems to have a battery contact problem - I have to check it) lost 6% and the worst case is a drop of almost 10%.

pvvx, 2023-12-27:
[TS0201_TZ3000](https://pvvx.github.io/TS0201_TZ3000/)
Works simultaneously in [BLE and Zigbee](https://github.com/pvvx/BZdevice). The cheapest 2 AAA batteries.
[image]

Neo2SHYAlien, 2023-12-27:
> The installation of two capacitors is provided. С24 and С25. 

Sorry for the silly question but what is recommended value for both capacitors ?

pvvx, 2023-12-27:
4.7..47 uF

tristone-cz, 2023-12-30:
Just an observation with Varta baterie, no noname Chinese ones.
[image]

Version 1.6
[image]

slingel, 2024-01-04:
with 0.1.1.8 no more battery drain for me.

muzzy124, 2024-01-09:
I got about 25-30% drain over 9 days with three LYWSD03MMC B2.0 flashed to 1141-020a-01183001-Z03MMC.zigbee
LQI (in ZHA) is about 70-120
[image]

pvvx, 2024-01-09:
I have Xiaomi LYWSD03MMC for 2 months now on an old battery previously used in the BLE version. All versions of Zigbee and other debugging options were flashed. I tried to turn off the coordinator for a couple of days (the biggest consumption when searching for a coordinator).
The battery two months ago was 60% - now 54%.

muzzy124, 2024-01-09:
i got ZHA with sonoff usb dongle cc2652 and a bunch of noname tuya relays and plugs

tristone-cz, 2024-01-09:
Me few cents guys.
I switched my B1.6 to "0.1.1.8" 5 days ago and it looks better compared to "0.1.1.6". Too early to conclude, the graph is bit fluctuating, but my estimation is about 1 month to get to 0%.
So better, but still problematic. Maybe the B1.6 is somehow weird?

pvvx, 2024-01-09:
One of the test groups of thermometers with Zigbee firmware.
[image]
For 2 months no problems with batteries.

Except Xiaomi Aqara:
[image]

pvvx, 2024-01-09:
> Maybe the B1.6 is somehow weird?

I ordered LYWSD03MMC from several other stores. I hope to get B1.6. Sent by mail. I'll check it when they arrive.
I don't know what to do with the rest of the accumulated LYWSD03MMC...

NorDroN, 2024-01-09:
I can send my LYWSD03MMC HW:B1.5 device from Russia for testing.

tristone-cz, 2024-01-09:
I had another two B1.6 so flashed to ATC 4.6 and placed to the same location at those Zigbee ones. Let's compare.

pvvx, 2024-01-14:
Test [Xiaomi LYWSD03MMC B1.5](https://github.com/pvvx/ATC_MiThermometer/tree/master/BoardPinout#xiaomi-lywsd03mmc-b15).
**Xiaomi's original BLE firmware** has an average consumption of **23 µA**.
BLE firmware **ATC_v46.bin** at default settings consumes less than **17 µA**.
Firmware **Zigbee 1141-020a-02043001-Z03MMC.zigbee**:
[image]
[Measurement with the settings specified in this issues.](https://github.com/pvvx/ZigbeeTLc/issues/6)

Xiaomi LYWSD03MMC B1.5 has the highest consumption in sleep mode (> 8 µA) and when polling the sensor.
The remaining versions (V1.4, V1.6, V1.7, V1.9, V2.0) have lower consumption.

In ZigBee mode, consumption depends on the coordinator. If it does not know how to work in Zigbey 3.0 or is terribly slow, then the consumption will be terrible.

tristone-cz, 2024-01-14:
Probably my last note.

The 0.1.1.8 version provides significant difference compared to 0.1.1.6.
BT sensors in graph are on ATC 46, the other two at Zigbee 0.1.1.8. Located in the same place, 1,5m from coordinator.
[image]

pvvx, 2024-01-14:
Versions below 0.1.1.8 have all interfaces: Touchlink commissioning, ZCL_GROUP.

muzzy124, 2024-01-23:
[image]
Как тебе такое, Илон Маск? :)
I've found that my sonoff-p coordinator firmware was rather old (2022***), so I upgraded it to the latest 20230507 on Jan 17th.
Afterwards it's been a kind of magic :) Battery level of two devices (orange and blue ones) began to go up.

pvvx, 2024-01-24:
Уже давно понятно, что c sonoff всё плохо.

drwalr, 2024-01-29:
It is possible to insert a larger battery into this sensor, it will take a CR2450 battery

[image]

https://www.olx.pl/d/oferta/wieksza-bateria-cr2450-do-xiaomi-lywsd03mmc-zigbee-CID99-IDYmMwB.html?bs=olx_pro_listing

slingel, 2024-02-01:
my comment from December:

> I'm seeing around 1mA with all your firmwares after interview with Sonoff CC2652P dongle on all my LYWSD03MMC. Every few seconds current drops below 10µA then right back to 1mA.

v1.1.1.7:
[image]

v1.1.1.9:
[image]

pvvx, 2024-02-01:
@slingel - what version (B1.4..B2.0) of the thermometer?

slingel, 2024-02-01:
1.6

zombie4000, 2024-02-08:
Just to add to this.
Device: LYWSD03MMC
This was my battery percentage on firmware 0x01183001 with a brand new battery, the same was happening with the included battery.
[image]

I upgraded to firmware 0x01193001 and its looking a lot more promising after 2 days:
[image]
 
On the earlier version the battery would be around 72-75% after 2 days. This was consistent between 3 different devices with included batteries and new batteries in all 3 devices. After all three devices behaved the same same with two sets of batteries I stopped using them until I upgraded one of them.

I can provide an update after a week.

[15 more replies]