pvvx/ZigbeeTLc #43: OTA в Zigbee2Mqtt

almirus, 2024-01-12:
Добрый день!
Это глюк, поддержки ОТА нет?
[image]
[image]

pvvx, 2024-01-12:
В чем глюк?
18362369 (dec)= 0x01183001 (hex)

almirus, 2024-01-12:
Последняя версия минус один

lduda007, 2024-01-14:
I have the same issue, -1 as the Latest version on Z2M 1.35.1

pvvx, 2024-01-14:
I do not use Z2M, and I do not publish firmware to general lists while they are in beta.

BenJamesAndo, 2024-01-15:
It's a confirmed issue with zigbee2mqtt and has been fixed in the dev branch.
https://github.com/Koenkk/zigbee2mqtt/issues/20658

kwiateusz, 2024-02-06:
Create custom OTA index https://www.zigbee2mqtt.io/guide/usage/ota_updates.html with payload:
`[
    {
        "fileVersion": 18427905,
        "fileSize": 128034,
        "manufacturerCode": 4417,
        "imageType": 522,
        "sha512": "f6310b3028dc0187ab99c6086368e467f3577101c6f8f1becdbfafcc22dcc1e1854ceddfd65504a45aef00a81b74530ba6425dcc41ee4b36a2d24426a1af3426",
        "url": "https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-020a-01193001-Z03MMC.zigbee"
    }
]`