pvvx/ZigbeeTLc #48: LYWSD03MMC: devbis -> pvvx wirelessly?

booroondook-eng, 2024-02-01:
Имеется 8 датчиков LYWSD03MMC, в свое время прошитых в Zigbee прошивкой отсюда: https://github.com/devbis/z03mmc
Можно ли установить на них прошивку из этого проекта через OTA?
Судя по принятой индексации, OTA-файл должен называться 1141-0203-... и т.д. с расширением .zigbee
Но подобного файла в репозитории я не нашел. Нашел только файл [Z03MMC_v0119.bin](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/Z03MMC_v0119.bin), предназначенный для прошивки через COM-порт (один датчик я прошил, но разбирать и паять провода к остальным очень не хотелось бы).
Спасибо!

pvvx, 2024-02-01:
https://github.com/pvvx/ZigbeeTLc/issues/33#issuecomment-1919404597

booroondook-eng, 2024-02-01:
То есть, либо через паяльник, либо через временный возврат на BLE?

pvvx, 2024-02-02:
Или сами делайте заголовочный файл для *.zigbee.

pvvx, 2024-02-02:
Пробуйте https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-0201-09993001-Z03MMC_v0119.zigbee

fhempy, 2024-02-04:
> Try https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-0203-09993001-Z03MMC_v0119.zigbee

Many thanks! That one makes the transition to your firmware an easy process. I updated it with zigbee2mqtt.

maaad, 2024-02-05:
> Пробуйте https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-0203-09993001-Z03MMC_v0119.zigbee
в последнем релизе сменился manufacturerID на 56085 и OTA ругается на Manufacturer code mismatch

pvvx, 2024-02-05:
python3 [zb_bin_ota.py](https://github.com/pvvx/ZigbeeTLc/raw/master/make/zb_bin_ota.py) [Z03MMC_v0118.bin](https://github.com/pvvx/ZigbeeTLc/raw/master/bin/Z03MMC_v0119.bin) Z03MMC_new -m 56085  -i0x0203 -v0x09993001 -s"Zigbee ver:devis to ZigbeeTLc"

pvvx, 2024-02-13:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis

ferrets6, 2024-02-20:
Thanks! 
Finally I successfully updated OTA on Zigbee2MQTT from devbis 1100 to pvvx v0120 using [this zigbee ota file](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/devbis/db15-0203-79993001-Z03MMC_v0120.zigbee)