pvvx/ZigbeeTLc #49: LYWSD03MMC: Zigbee2MQTT: Device left

booroondook-eng, 2024-02-02:
Не могу зарегистрировать перепрошитый LYWSD03MMC в сети Zigbee2MQTT. Ситуация такая. Была прошивка от devbis.
Удалил датчик из сети (принудительно). Прошил Z03MMC_v0119.bin (по проводам). Запускаю в сети join - устройство "возникает", начинается его опрос. Все как обычно, доходит до появления сообщения "устройство успешно прошло интервью" (точный текст на английском не помню, но суть понятна), и... через пару секунд сообщение "Device такое-то left the network".
Уже раз десять пробовал - и зажимать сброс не 7 секунд, и просто передергивать батарейку, и перезапускать службу Z2M. Но результат один и тот же - в самый последний момент, когда казалось бы, все настройки закончены, происходит этот "выход" из сети.

pvvx, 2024-02-02:
Забыл главное - см п.п.7 [README](https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#to-flash-the-zigbee-firmware-use-a-google-chrome-microsoft-edge-or-opera-browser):
_Если этого не произошло, вставьте аккумулятор заново и/или закоротите контакты RESET и GND на плате LYWSD03MMC, а на датчиках с кнопкой нажмите кнопку и удерживайте ее 7..8 секунд._

pvvx, 2024-02-03:
Почему в ZHA ничего не отваливается и всё работает, а в z2m все так плохо?

> "Отличный" совет - ради работы пары датчиков, поменять всю систему. Это примерно, как поменять автомобиль из-за неработающей лампочки. Особенно, когда в этой сети прекрасно работает три десятка устройств разных производителей.

По причине что с Sonoff больше всего проблем и у пользователей.
Возможно что Sonoff работает только с теми устройствами, для которых его адаптировали. Но не работает с другими.
Это как с z2m и ZHA - пока вам его создатели не разрешат работать с другим типовым устройством Zigbee 3.0, ничего работать нормально не будет.

booroondook-eng, 2024-02-04:
> По причине что с Sonoff больше всего проблем и у пользователей.
>
Посоветуйте альтернативу. Желательно, с такой же дальнобойностью, как у Sonoff Dongle-E

closed 2025-11-07 by pvvx