pvvx/ZigbeeTLc #51: LYWSD03MMC B1.6 device works fine with BT but not Zigbee

angelsl, 2024-02-04:
I have 4 LYWSD03MMC B1.6s. Three of them work fine with this Zigbee firmware. But one of them simply refuses to join my Zigbee network (using ZHA). It works fine when I flash (by UART) the custom BT firmware (from https://github.com/pvvx/ATC_MiThermometer).

I have no idea how to debug this.. Any advice appreciated. Thanks.

tell-1c, 2024-02-05:
I encountered that the device did not work when battery power low at 2.8-2.85V. Try installing a new battery.

kristoficko, 2024-03-20:
@angelsl same here, out of 7 devices I have one that works with BT, but no matter what zigbee firmware and how many times I flash (OTA or UART, devbis or pvvx) it just won't connect to my zigbee controller. Other 6 devices work flawlessly. Tried with multiple new batteries (I switched to Duracell as the random chinese batteries are very weak and unstable in voltage).

pvvx, 2024-03-20:
Check the MAC address. 3 MAC bytes are randomly selected.
You can change the MAC in the BLE firmware.
After erasing all Flash, a new MAC will be set.
Flash has correction records for quartz and others. And if they are knocked down, it is better to erase all Flash and then write a new program.

kristoficko, 2024-03-21:
Thanks for pointing this out, unfortunately I'm not sure it applies here - for UART flashing I'm using a RPi GPIO TTL and your python programming script TLSR825xComFlasher.py.

Everytime I flash a thermometer through UART I first choose 'ea' - erase all function and afterwards I write the firmware. It is true that it helped me resolve pairing issues with few of the already working devices, but this one still resists. Already flashed ~10 times with ea first.

closed 2025-11-07 by pvvx