pvvx/ZigbeeTLc #57: Where is BLE Zigbee OTA for TS0201 tuya?

khayamgondal, 2024-02-10:
OTA on the following link doesn't contain one for TS0201. I am trying to go back to BLE from Zigbee
https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

robvanoostenrijk, 2024-02-10:
Its there on the dedicated repo for the TS0201:

https://github.com/pvvx/BLE_THSensor
> Firmware file for recovery from Zigbee to BLE: [1141-0211-99993001-TS0201_v45.zigbee] (https://github.com/pvvx/BLE_THSensor/raw/master/source/TS0201/bin/1141-0211-99993001-TS0201_v45.zigbee)

pvvx, 2024-02-11:
Initially, TS201 is flashed by the programmer through the contacts of the SWS and RST connector or using the Zigbee OTA file [1141-d3a3-00993001-TS0201_v45.zigbee](https://github.com/pvvx/BLE_THSensor/raw/master/source/TS0201/bin/1141-d3a3-00993001-TS0201_v45.zigbee).

closed 2025-11-07 by pvvx