pvvx/ZigbeeTLc #63: No Zigbee Firmware flash offered

nils-eman, 2024-02-17:
Hi! 
I was able to flash custom firmware 4.6 OTA on my MHO-C401, but after the reconnect I only get the option for Custom 4.6 or Original_Xiaomi OTA to flash. Even when trying to choose a firmware manualy I get an "incorrect head in Telink OTA binary firmware" error.
Tried two PCs and a mobile - always the same....
Any advice?

pvvx, 2024-02-17:
[README](https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#custom-firmware-for-zigbee-30-iot-devices-on-the-tlsr825x-chip)
_In developing: [MHO-C401(old)](https://pvvx.github.io/MHO_C401), [MJWSD05MMC](https://pvvx.github.io/MJWSD05MMC), [CGG1-M](https://pvvx.github.io/CGG1), [TS0202_TZ3000](https://pvvx.github.io/TS0202_TZ3000)_

nils-eman, 2024-02-18:
Hmm, according to the specs on the packaging it matches the MHO-C401N specs, but however. Thanks for the good work, let's see when the old version will be supported

elektrinis, 2024-07-13:
How to tell them apart when ordering? Do they look different?
I think many sellers are still selling C401?

closed 2025-11-07 by pvvx