pvvx/ZigbeeTLc #64: lywsd03mmc all characters on screen

gordonpt, 2024-02-18:
Recently 3 of my thermometers became like the image attached.

Already tried to reset, any ideas of what could be happening?

Thanks in advance,
Cheers 
[image]

nonflammable, 2024-02-19:
try to replace the battery with a new one

lwasyl, 2024-04-03:
Same issue here. I'm not sure if it happened after update to `3001-0118` or prior. The battery has charge left and replacing the battery with one from a working unit doesn't help. The device is also offline in Zigbee2MQTT

nonflammable, 2024-04-03:
>The battery has charge left and replacing the battery with one from a working unit
try NEW one