pvvx/ZigbeeTLc #65: MHO-C401 brick my devices

Stawrit, 2024-02-19:
I guess I didn't see the N at the end of the model number. Did I break my devices?

[image]
[image]

Stawrit, 2024-02-22:
I restored the original firmware of the device.  Are you still working on zigbee compatible firmware for this device?  I have 7 of them and I don't want them to go to waste.

Dominiator, 2024-02-27:
Это из-за  того что помеялось имя устройства, а в Zigbee2MQTT конвертер только для MHO-C401N, а не MHO-C401N-z. Установка конвертера вручную спасло ситуацию

willumpie82, 2024-02-28:
I did the following:
flashed custom 46 firmware OTA (chrome windows) = OK 
flashed zigbee OTA (chrome windows) = Not OK ==> bricked?
tried to flash with serial FullFlash_MHO-C401_v1.0.0_0010.bin ( 524KB) (chrome Macos)= Not OK
tried to flash with serial Original_OTA_Xiaomi_MHO_C401_v1.0.0_0016.bin (chrome Macos) = Not OK
see below my connection, when 'sending firmware' the TX led is blinking so 'stuff' is happening
[image]

can some one help me out (pref in English ;-) )

pvvx, 2024-02-28:
> Это из-за того что поменялось имя устройства, а в Zigbee2MQTT конвертер только для MHO-C401N, а не MHO-C401N-z. Установка конвертера вручную спасло ситуацию

[А это разве не работает](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0)? 

> This release contains a quite fundamental change on how Zigbee2MQTT handles unsupported devices.
>Zigbee2MQTT will now attempt to automatically detect features supported by the device and discover those to e.g. the frontend and Home Assistant.
>This means that if you, for example pair an unsupported light, there is no need to wait for official support. Basic features will work out-of-the-box.

До сих пор всё настолько криво в z2m?

willumpie82, 2024-02-28:
I haven't been able to connect to zigbee, it bricked after installation of OTA firmware (manual installation, in online flasher the 'zigbee firmware' button did not apear)

but anyway i'm using latest z2m (ha add-on)

Dominiator, 2024-02-28:
> [А это разве не работает](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0)?
>
>
> До сих пор всё настолько криво в z2m?

Он показал только батарейку, температуру, влажность(остальные параметры и настройки были недоступны), я минут 10 ждал, но датчик не обновлял данные в z2m. Скопировал конвертер MHO-C401N, переименовал MHO-C401N-z и сразу все данные стали прилетать. А так все отлично, спасибо за вашу работу!

closed 2025-11-07 by pvvx