pvvx/ZigbeeTLc #67: IH-K009 support required

KonsGit, 2024-02-25:
This is a zigbee temperature and humidity sensor. Look the same as TH03Z. The sensor used is SHT30 sensor.
I tried to flash this sensor with the firmware from TS0201_TZ3000, but it doesn’t show anything except the battery. Temperature and humidity are shown as 0.
I tried to flash the BLE version from TH03Z and it also shows 0 temperature and humidity.
It would be great if your firmware supported these sensors.
[image]
[image]
[image]

10Silver10, 2024-02-25:
Good afternoon. I am also interested in this topic.

pvvx, 2024-02-25:
What numbers does Zigbee OTA have for the original Tuya version?
Manufacturer Code | Image Type | File Version
-- | -- | --
| ? | ? | ? |

KonsGit, 2024-02-26:
Now I can’t look up these values in zha.
In zigbee2mqtt I see the following information for sensor with original firmware (database.db file):

{"id":23,"type":"EndDevice","ieeeAddr":"0xa4c138bf7fbb7257","nwkAddr":28236,"manufId":4417,"manufName":"_TZ3000_dowj6gyi","powerSource":"Battery" ,"modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[ 1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":-1998}},"msRelativeHumidity" :{"attributes":{"measuredValue":6567}},"genBasic":{"attributes":{"65503":"�|Y-\u0011�[-\u0012\u0001\u0000\u0000\u0000\ u0011","65506":51,"65508":1,"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_dowj6gyi","powerSource":3,"zclVersion":3," appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":2,"batteryVoltage":25}}} ,"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""," zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1708668821325,"defaultSendRequestWhen":"immediate"}

I.e. 
manufId:4417
manufName:_TZ3000_dowj6gyi
modelId:TS0201

pvvx, 2024-02-26:
> The sensor that I currently have has already been converted to ble and shows zero temperature and humidity

https://github.com/pvvx/BLE_THSensor/blob/master/source/TS0201/bin/TH03Z_v45.bin - shows zero temperature?

Zigbee + BLE for HT03Z https://github.com/pvvx/BZdevice/tree/master/bin

---

Z2M and ZHA hide information about Zigbee devices.
To get information about the firmware in the device, you must enable debug log and run the OTA update command, as shown here:
https://github.com/pvvx/ZigbeeTLc/issues/7#issue-2000695703

KonsGit, 2024-02-27:
Yes, is s shows zero
[image]
[image]

pvvx, 2024-02-27:
For testing, I flashed the specified file:
https://github.com/pvvx/BLE_THSensor/blob/master/source/TS0201/bin/TH03Z_v45.bin 

There are discrepancies:
```
09:41:59: File: TH03Z_v45.bin
09:41:59: File size: 75316 bytes
09:41:59: Count: 4708
....
09:43:58: Hardware Revision String: V1.0
09:43:58: Software Revision String: V4.5
09:43:58: Firmware Revision String: github.com/pvvx
09:43:58: Detected custom Firmware
09:43:58: Hardware Version: TH03Z V1.0, Software Version: 4.5, Sensor: GXHT30
```

KonsGit, 2024-02-27:
still the same
[image]
[image]

KonsGit, 2024-03-03:
"fileVersion":69,
"imageType":54179,
"manufacturerCode":4417

KonsGit, 2024-03-05:
> What numbers does Zigbee OTA have for the original Tuya version?
> 
> Manufacturer Code	Image Type	File Version
> ?	?	?

From sensor with original firmware:

fileVersion:69,
imageType:54179,
manufacturerCode:4417

pvvx, 2024-03-05:
fileVersion: 69 = 0x45,
imageType: 54179 = 0xD3A3,
manufacturerCode: 4417 = 0x1141 (Telink)
[1141-d3a3-00993001-TH03Z_v45.zigbee](https://github.com/pvvx/BLE_THSensor/raw/master/source/TS0201/bin/1141-d3a3-00993001-TH03Z_v45.zigbee) оriginal -> [BLE_THSensor](https://github.com/pvvx/BLE_THSensor) -> [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) + any firmware.

KonsGit, 2024-03-07:
This firmware does not work on my sensor.
Look at the pictures above.
It shows 0 temperature and the sensor defines it as CHT8305, which is not correct. I have SHT30

pvvx, 2024-03-07:
https://pvvx.github.io/TH03Z/

KonsGit, 2024-03-07:
Thank you very much
Latest version bin/ZTH03Z_v0120.bin works great!

10Silver10, 2024-03-09:
> Thank you very much Latest version bin/ZTH03Z_v0120.bin works great!

Hello. You can describe the firmware process in more detail. Thank you in advance.

closed 2024-04-15 by KonsGit