pvvx/ZigbeeTLc #68: Bricked device TS0201_TZ3000 Zigbee

andreasdijkman, 2024-03-02:
I tried flashing this firmware with [WebUSBCOMflasher](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) over the default Tuya firmware, but now it doesn't show up anywhere. I tried using the BLE-firmware from [TS0201_v45.bin](https://github.com/pvvx/BLE_THSensor/raw/master/source/TS0201/bin/TS0201_v45.bin) and the Zigbee firmware.bin. Both to no success (I guess).

After flashing the red LED doesn't turn on, which it did on the original firmware. Pressing the button or shorting RST to GND for 7 seconds doesn't do anything either. Also in the BLE-scan with the BLE-firmware (hopefully) it doesn't show up.

I have a default FTDI Programmer set to 3.3v. I hooked GND + SWS-TX but it doesn't do anything.

Even trying the [TlsrComProg825x](https://github.com/pvvx/TlsrComProg825x) didn't work. Any diagnostic steps or special instructions of recovering from a bad flash or watching some output on some TX/RX-lines?

Help would be greatly appreciated.

robvanoostenrijk, 2024-03-02:
@andreasdijkman,

There is a resolved issue in the dedicated repo for TS0201_TZ3000 which might more information for you:
https://github.com/pvvx/BLE_THSensor/issues/7

andreasdijkman, 2024-03-03:
> @andreasdijkman,
> 
> There is a resolved issue in the dedicated repo for TS0201_TZ3000 which might more information for you: [pvvx/BLE_THSensor#7](https://github.com/pvvx/BLE_THSensor/issues/7)

Thanks! I didn't see the issue bit it looks like it's exactly the same as I have. @robvanoostenrijk how did you confirm that the flash was succesfull? Did the red LED came to life or did the button trigger the red flash? Or did it only show up as a `THS_XXXX` device in the bluetooth flasher?

robvanoostenrijk, 2024-03-04:
After flashing [TS0201_v45.bin](https://github.com/pvvx/BLE_THSensor/raw/master/source/TS0201/bin/TS0201_v45.bin) with [WebUSBCOMflasher](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html), the red LED should work when pressing the reset button.

As mentioned in the closed issue, after flashing you might have to leave the device off power for an hour or so.
Only then the Bluetooth showed for me, the red light was always working after succesful flash though (once I had the wires correct as per the feedback from Webbrain).

andreasdijkman, 2024-03-04:
Is there some debug console output on the TX/RX pins of the MCU of somewhere else? I tried 2 different USB-TTL-flashers with the WebUSBCOMFlasher on macOS with Google Chrome. Nothing... 

I'm now waiting on the flasher that is mentioned in the other issue but Europe is a long way from China and expected in 14 days...

pvvx, 2024-03-04:
TS0201_TZ3000 is flashed via Zigbee OTA using the file [1141-d3a3-00993001-TS0201_v45.zigbee](https://github.com/pvvx/BLE_THSensor/raw/master/source/TS0201/bin/1141-d3a3-00993001-TS0201_v45.zigbee)

andreasdijkman, 2024-03-04:
Yes, I know now, but it's too late for that for 1 of my sensors...

I'm having problems flashing by ISP and I can't verify if the flash is succeeded as I don't see any output if the flash succeeded or not.

andreasdijkman, 2024-03-04:
Thx, I'll try my 2nd adapter. 

Looking better at the TlsrComSwireWriter-readme is clearly says:
> *  Does not work on USB-COM adapters that have LEDs on at the RX input.
> *  Does not work on USB-COM adapters that have FTDI chip

Guess what my first adapter is and has, and FTDI-chip with LED's on both TX and RX. I'll try my ESP-01-flasher with this program.

andreasdijkman, 2024-06-10:
I'm still struggling with the devices. Is there a way to check if my device is still alive? I don't now if it is completely dead or just bricked. Some kind of uboot-output on pins or something?

web2brain, 2024-08-24:
After I struggled to revive my bricked TS0201 from https://github.com/pvvx/BLE_THSensor/issues/7, I was successfull with a different approach than the one @robvanoostenrijk used: I directly flashed the Zigbee firmware https://github.com/pvvx/ZigbeeTLc/blob/master/bin/ZTS0201Z3000_v0122.bin to the device, skipping the BLE firmware. This worked like a charm.
@robvanoostenrijk I finally got my own TTL flasher :)