pvvx/ZigbeeTLc #69: Victor, I need to talk to you.

ysoyipek, 2024-03-03:
I couldn't find his email address anywhere. I apologize for writing here. After reading this we can delete it immediately. I don't want to dirty this place. I want to talk business with you about TLSR8258 Zigbee. Can you please reach me? yusuf@soyipek.com

pvvx, 2024-03-04:
I communicate only in open public sources.

ysoyipek, 2024-03-04:
> I communicate only in open public sources.

Thank you for the answer. I came across this project for a Zigbee sensor designed with TLSR8258. I am asking for help in return for a fee or donation. Please try to understand me. Since it will be a commercial product, I have to prefer one-to-one communication instead of public. I also think I should pay a royalty when I use your code and make money from it.

pvvx, 2024-03-04:
The [license ](https://github.com/pvvx/ZigbeeTLc/blob/master/LICENSE) does not imply any copyrights or restrictions. This is not GPL.
The Telink SDK code has its own license.

ysoyipek, 2024-03-04:
> The [license ](https://github.com/pvvx/ZigbeeTLc/blob/master/LICENSE) does not imply any copyrights or restrictions. This is not GPL. The Telink SDK code has its own license.

I am aware, yes the license does not include royalty. But you contributed to this and shared it without any expectations. This is something that needs to be appreciated.

Also, from a professional perspective, you are more familiar with the source code than I am. I ask for help if you deem it appropriate. And when you provide this assistance, according to my ethics, there should be a reward for it.

pvvx, 2024-03-04:
When copying, redistributing and using, you do not need to credit me as the developer or provide any links to my code. Only if you want to do it yourself. That is, there are no copyright restrictions.

I don't have much time to help with other developments. And I am not a professional programmer, since I am developing small electronic devices with microcontrollers.

ysoyipek, 2024-03-04:
> When copying, redistributing and using, you do not need to credit me as the developer or provide any links to my code. Only if you want to do it yourself. That is, there are no copyright restrictions.
> 
> I don't have much time to help with other developments. And I'm not a professional programmer, as I've been developing small electronic devices with microcontrollers since 1985. And the age is no longer the right one to engage in small-scale commerce. I have had my own production in the industrial sector for over 40 years...
> 
> And I don’t know English and don’t want to learn it. I use google translator.

I understand you, but thank you for replying to my message.

closed 2024-03-04 by ysoyipek