pvvx/ZigbeeTLc #70: LYWSD03MMC first char not visible

naimadfpv, 2024-03-06:
characters on screen are not visible at all.

tried with a fresh new battery and another lcd screen but always the same as in picture.

[image]

KennethLavrsen, 2024-03-13:
I bet this is a poor connection between PCB and the LCD. The connection is done using a zebrastrip. Take it apart, clean the pcb, the two surfaces, and the lcd with isopropylic alcohol. Be very gentle with the zebra strip. And when you reassemble make sure the PCB is clicked all the way down and the plastic frame is secured by the two screws. It is impossible that the software randomly screws up missing segments unless it happens to everyone.

naimadfpv, 2024-03-13:
I thought that this could happen due to a bad connection to the lcd and tried to clean the pcb. 
I didn’t found contacts on the lcd so I couldn’t clean it.

Never heard of a zebrastrip before. so I can disassemble the zebrastrip from the lcd and clean the contacts of the lcd? Or did I missunderstood something?

KennethLavrsen, 2024-03-13:
HiIf the zebrastrip is still fully attached to the LCD then it is probably not dirty. The zebra strip is a fantastic invention. The black stripe in the center is actually tiny small carbon strips isolated from each other. Each terminal of PCB and LCD is covered by maybe 10 small channels. This means that the zebra strip does not need to be alligned very well. As long as the LCD and PCB is alligned it is OK. The zebra strip is not glued to the LCD. It is just pressed hard against and the silicone is a bit sticky. But you can carefully remove it and clean it. Carefully. The metal contacts on the glass are only a few atoms thick and transparent. The key thing for the assembly to work is that there is an even pressure on the entire length. Make sure the two screws are tight but do not over tighten. Careful is the key. I took 4 of these apart and flashed them with serial interface and had these missing segment on several until I had cleaned and properly pressed the frame down and it was not until the last screw was tight that the segments were all on. One I had to clean.I have also designed LCD assemblies as an engineer years ago which is why I know these zebrastrips well. Zebra because of the tiny little stripes you see in a microscope.Kenneth

naimadfpv, 2024-03-14:
Thank you for the clearly and detailed info about zebrastrips. 

I switched the lcd including the zebrastrip with one that is working correctly in another sensor and that lcd has also missing signs at the same position so maybe it’s the pcb and not the lcd or zebrastrip? 

But I will give it a try with disassembling the zebrastrip and cleaning the lcd and pcb.

pvvx, 2024-03-14:
First you need to reflash the device.
You did not describe how you flashed the thermometer and there may be errors in Flash. There are many reasons - low battery voltage when flashing via OTA, poor contact when flashing via UART, ...

naimadfpv, 2024-03-15:
Sorry I forgot to say that.

So I first flashed Firmware with your OTA Updater (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) to Firmware 4.6. 
Then to Zigbee Firmware 0119. Everything worked correctly.
After 1-2 weeks I updated Firmware with z2m to new zigbee-firmware 0120 and the device worked again for some weeks (after i set the binding correct as mentioned here in an closed issue i cannot find yet).
One or Two Days before i raised that issue I saw that one char on the lcd was missing so i decided to have a look inside the thermometer and cleaned the contacts of the pcb where it is connected to the lcd (zebrastrip).

I had to switched the battery because i reconnected it very often so the battery was emtpy. But same issue with a brand new battery.