pvvx/ZigbeeTLc #71: Device didn't respond to OTA request

nchieffo, 2024-03-06:
Running on Z2M and an update notification arrived (somebody listed this repo in the Z2M OTA index marked with version 18886657)

This is what I see after 1-2 mins

`2024-03-06 19:27:51Update of 'MiTemp Ingresso' failed (OTA: Device didn't respond to OTA request)`

Tried multiple times without success

My LYWSD03MMC is currently on version 3001-0118

zoilomora, 2024-06-02:
The same happens to me.

pvvx, 2024-06-03:
https://github.com/pvvx/ZigbeeTLc/blob/master/bin/index.json
```
    {
        "fileVersion": 18952193,
        "fileSize": 127298,
        "manufacturerCode": 4417,
        "imageType": 522,
        "sha512": "7dbb388c19b2d80d184069243141771f938d560b16ca41b1cfc6daf9151ddb2038becdec8010ee2b24763ebdb92f96efb54f51f0e81764ab3e399a52a43c6725",
        "url": "https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-020a-01213001-Z03MMC.zigbee"
    },
```
[zigbeetlc_v0121.js](https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/zigbeetlc_v0121.js)

zoilomora, 2024-06-03:
@pvvx I have been able to forcefully update the firmware.

Now without having any external converters it detects the device name `LYWSD03MMC-z`:

[image]

Thank you very much!