pvvx/ZigbeeTLc #74: LYWSD03MMC schematic

tovy14, 2024-03-15:
Hi,
After flashing the 1st sensor, i thought i would do them all, boy i was wrong. I managed to rip the P14 pad off the board while soldering.

Does someone have a schematic for where the P14 pad is connected to on the board?

Thank you.

pvvx, 2024-03-15:
https://github.com/pvvx/ATC_MiThermometer/tree/master/BoardPinout

P14 -> SWS ->  pin N4 TLSR8251F512ET24

[image]

tovy14, 2024-03-15:
> https://github.com/pvvx/ATC_MiThermometer/tree/master/BoardPinout
>
> P14 -> SWS -> pin N4 TLSR8251F512ET24 [...]

Thank you very much, sorry still kind of new top Github, will check better next time!

closed 2024-03-15 by tovy14