pvvx/ZigbeeTLc #76: Большой период обновлений Яндекс Станция

Go2gagra, 2024-03-22:
Здравствуйте! 
Датчик на последней прошивке из вашего репозитория присылает показания нерегулярно, влажность примерно раз в 10 минут, а температура вообще по полчаса, и больше. 
раньше на прошивке от devbis 1.0.6 показания приходили четко раз в минуту, но датчик отваливается если выключить хаб (станция макс) на некоторое время.

Go2gagra, 2024-03-22:
При замыкании  ground-reset данные прилетают моментально

pvvx, 2024-03-22:
Интервалы передачи в Zigbee задаются координатором при связывании кластеров...

pvvx, 2024-03-23:
> Датчик на последней прошивке из вашего репозитория присылает показания нерегулярно, влажность примерно раз в 10 минут, а температура вообще по полчаса, и больше.

Видимо так назначает reporting "Яндекс Станция". Берегут свой сервер от частых передач и обновлений логов. :)
Пишите им, чтобы сделали настройки нормальными для работы датчиков "Умного дома". А это интервалы для термометров 10..30 секунд, если они управляют обогревом или кондиционером и т.д. :P

Go2gagra, 2024-03-23:
А могу я сам собрать прошивку, где можно принудительно указать интервал как было на 1.0.6 от девбис?

pvvx, 2024-03-23:
А нет такого - принудительно. Это будет не соответствовать Zigbee 3.