pvvx/ZigbeeTLc #77: LYWSD03MMC-z temp/hum is null

ruvasik, 2024-03-23:
```
Last seen N/A
Availability [Disabled]
Device type EndDevice
Zigbee Model LYWSD03MMC-z
Zigbee Manufacturer Xiaomi
Support status Not supported
IEEE Address 0xa4c1386b31feea57
Network address 0xA87B
Firmware build date 20240226
Firmware version 3001-0120
Power Battery 100%
Interview completed True

Battery
Remaining battery in % 100%
Temperature Measured temperature value N/A°C
Humidity Measured relative humidity N/A%
Linkquality Link quality (signal strength) 23lqi
```

In state i see only `{
    "battery": 100,
    "linkquality": 23
}`

On try change reporting interval
```
Bind 0xa4c1386b31feea57/1 msTemperatureMeasurement from '0x00124b001cd502f5/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c1386b31feea57/1 msTemperatureMeasurement from '0x00124b001cd502f5/1' failed (AREQ - ZDO - bindRsp after 10000ms)'
```

In HA logs i see not errors.

ruvasik, 2024-03-23:
In dev console i can read values
Humadity measuredValue 
Read result of 'msRelativeHumidity': {"measuredValue":4421}
Temperature measuredValue 
Read result of 'msTemperatureMeasurement': {"measuredValue":2284}

pvvx, 2024-03-23:
The clusters binding and the configuration report do not work for you.