pvvx/ZigbeeTLc #78: Can't connect Qingping Temp RH Lite to HomeAssistant

muscaiu, 2024-03-26:
I'm trying to add Qingping Temp RH Lite to HomeAssistant by flashing it with Telink Flasher, but it gets stuck on this step "Activating now, please wait..."

[image]

I have other Xiaomi devices added with this method, but no luck with this first QingPing device i just bought.

[image]

Need to mention i don't have a Xiaomi hub, i'm using USB dongle attached to RPi.    
Any suggestion?

robvanoostenrijk, 2024-03-26:
You need to press the button on the back of the Qingping while doing the activation.
If you get the timing right, it will go through eventually.

muscaiu, 2024-03-26:
I just realized it works if you keep the back button pressed for 5 seconds. /facepalm.
Closing

closed 2024-03-26 by muscaiu