pvvx/ZigbeeTLc #79: Qingping Zigbee wrong readings in HomeAssistant

muscaiu, 2024-03-26:
I just flashed the Qingping Temp RH with the TelinkFlasher using ZIGBEE, added it to HomeAssistant, but the data readings in HA are always on 24.1temp and the humidity is kind of the same, just rarely spiking lagged somethimes.

[image]

I tried reconnecting, but the device is not visible in TelinkFlash anymore.

[image]

How can i revert and try the bluetooth version maybe that works well with HA?

robvanoostenrijk, 2024-03-26:
I use 3 of these with HA in ZigBee mode, firmware 3001-0120.

Are you using ZHA or Z2M?

pvvx, 2024-03-26:
> How can i revert and try the bluetooth version maybe that works well with HA?

The new version of ZHA has problems with OTA for many devices. I didn’t find out the reason. I think we need to wait for the new version of ZHA.

> I just flashed the Qingping Temp RH with the TelinkFlasher using ZIGBEE, added it to HomeAssistant, but the data readings in HA are always on 24.1temp and the humidity is kind of the same, just rarely spiking lagged somethimes.

By default, ZHA uses large "report" intervals.

Try using "reconfigure"

closed 2025-11-07 by pvvx