pvvx/ZigbeeTLc #80: increasing reporting frequency

TomW1605, 2024-03-28:
Hi, is there a way to increase how often the LYWSD03MMC reports temp and humidity? i want to use it in an AC control automation but it updates way too slowly for that to be practical. i understand this will increase battery usage but its just not useful to me as it is now so if we could get that as an option it would be great

pvvx, 2024-03-28:
For the Zigbee version there is no such option without changing the program. For BLE, configurable up to 62.5 ms.

TomW1605, 2024-03-28:
if i was to compile the firmware myself could i change this value? or is it a significant change?

pvvx, 2024-03-28:
Zigbee uses quarter second transaction steps. That is, if 3 reports were triggered, it takes 3*0.25 seconds to transmit them.

Min:  
#define READ_SENSOR_TIMER_SEC 	1
https://github.com/pvvx/ZigbeeTLc/blob/master/src/app_cfg.h#L85
#define DEFAULT_POLL_RATE					(4 * (4 * POLL_RATE_QUARTERSECONDS))
https://github.com/pvvx/ZigbeeTLc/blob/master/src/app_cfg.h#L139
if(rep_uptime_sec >= READ_SENSOR_TIMER_SEC) {
https://github.com/pvvx/ZigbeeTLc/blob/master/src/device.c#L273

As a result, the coordinator will not have time to receive other devices, the signal from which will be weaker. There will be a limitation on the number of devices processed.

TomW1605, 2024-03-29:
thanks for the info, i dont really need the screen in this room so i might just swap things around.

closed 2024-03-29 by TomW1605