pvvx/ZigbeeTLc #82: CGDK2: Missing ZigBee Cluster

robvanoostenrijk, 2024-04-03:
When using multiple CGDK2 devices, I found an interesting issue.

All of the devices are paired with Z2M and report their statistics successfully. 

However, one of the devices fails to report a `hvacUserInterfaceCfg` input cluster.

**Device with the issue:**
[image]

[image]

**Device working normally:**
[image]

[image]

Firmware on both devices was updated through OTA successfully and is identical. 
I've tried to remove & repair and reconfigure the problematic device, but it keeps reporting these exact clusters.

Any idea why this could be the case? Is there some configuration on the device itself which is causing the firmware to report different clusters?

Is there any way to reset the device local config / data, I assume this requires going to BLE firmware and back?

pvvx, 2024-04-03:
Definition for Incoming clusters:
https://github.com/pvvx/ZigbeeTLc/blob/master/src/sensorEpCfg.c#L67

ZCL_CLUSTER_HAVC_USER_INTERFACE_CONFIG is defined only in outClusterList:
https://github.com/pvvx/ZigbeeTLc/blob/master/src/sensorEpCfg.c#L89

robvanoostenrijk, 2024-04-07:
Thanks, I'll adjust the Z2M convertors to follow the in/out cluster definition as per source code.

closed 2024-04-07 by robvanoostenrijk

pvvx, 2024-04-07:
In the next version of HAVC USER_INTERFACE will be moved to "input clusters".
Previously, only when HAVC USER_INTERFACE was in the "output clusters" list did third-party software work correctly. Now it's fixed and needs to be done correctly.