pvvx/ZigbeeTLc #85: LYWSD03MMC-z Update from 1.20 to 1.21 "Boot Loop?"

BWFCGUIXT, 2024-04-18:
Had the following firmware installed: [Z03MMC_v0120.bin](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/Z03MMC_v0120.bin)

Now used the following file for OTA updates: [1141-020a-02053001-Z03MMC.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-020a-02053001-Z03MMC.zigbee)

After this all 5 of my devices are in the following loop:

[image]

BWFCGUIXT, 2024-04-18:
`19:22:48: Searching for devices
19:23:04: Connecting to: ATCzB5AA2C
19:23:09: Hardware Revision String: V0E
19:23:09: Software Revision String: Z0.2.0.5
19:23:09: Firmware Revision String: github.com/pvvx
19:23:09: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
19:23:09: Reconnect 1 from 5
19:23:14: Hardware Revision String: V0E
19:23:14: Firmware Revision String: github.com/pvvx
19:23:14: Disconnected.
19:23:14: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
19:23:14: Reconnect 2 from 5
19:23:27: Hardware Revision String: V0E
19:23:27: Software Revision String: Z0.2.0.5
19:23:27: Firmware Revision String: github.com/pvvx
19:23:27: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
19:23:27: Reconnect 3 from 5
19:23:27: Hardware Revision String: V0E
19:23:27: Software Revision String: Z0.2.0.5
19:23:27: Firmware Revision String: github.com/pvvx
19:23:27: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
19:23:27: Reconnect 4 from 5
19:23:27: Hardware Revision String: V0E
19:23:27: Software Revision String: Z0.2.0.5
19:23:27: Firmware Revision String: github.com/pvvx
19:23:27: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
19:23:27: Reconnect 5 from 5
19:23:28: Hardware Revision String: V0E
19:23:28: Software Revision String: Z0.2.0.5
19:23:28: Firmware Revision String: github.com/pvvx
19:23:28: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
19:23:28: Something went wrong, too many reconnects
19:23:32: Disconnected.`

BWFCGUIXT, 2024-04-18:
All devices are advertising over Bluetooth in this format : ATCzB5AA2C

But trying : https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

Fails connecting as seen in the log above.

bearklake, 2024-04-19:
Такая же ерунда. Как теперь восстановить датчик? Он как будто в вечном ребуте.

BWFCGUIXT, 2024-04-19:
So after a night on this status, I think they depleted the battery and are now all stuck on the following screen:
[image]

TeHashX, 2024-04-19:
Same here with 2 devices, I hope we can do OTA somehow and not USB flashing https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html

BWFCGUIXT, 2024-04-19:
I managed to get them showing upa as Zigbee devices by shorting GND + Reset for 7 seconds

[image]

1 Device shows up as LYWSD03MMC and the other as LYWSD03MMC-bz both of course not working in Zigbee2MQTT :)

BWFCGUIXT, 2024-04-19:
I'm fairly confident that all characters showing on the screen is due to a low battery.

I get them all connected into Home Assistant via BTHome and now half via Zigbee into Zigbee2MQTT.

The screen however still loops like in the GIF posted above and battery drain is crazy.

BWFCGUIXT, 2024-04-19:
After restarting and removing the converter, now 1 shows up -z and the other still -bz

[image]

BWFCGUIXT, 2024-04-19:
Just tried with ZHA in Home Assistant, it recognizes the device but also can't configure it 100% correctly.

BWFCGUIXT, 2024-04-19:
So getting it connected seems to work :) however trying for example the option for Hide Smiley:

`Zigbee2MQTT:error 2024-04-19 15:05:07: Publish 'set' 'show_smiley' to '0xa4c13836f6377592' failed: 'Error: ZCL command 0xa4c13836f6377592/1 hvacUserInterfaceCfg.write({"2":{"value":1,"type":48}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9283 - 1 - 69 - 516 - 4 after 10000ms)'
Zigbee2MQTT:debug 2024-04-19 15:05:07: Error: ZCL command 0xa4c13836f6377592/1 hvacUserInterfaceCfg.write({"2":{"value":1,"type":48}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 9283 - 1 - 69 - 516 - 4 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:362:9)
    at Object.toZigbee.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1431:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)`

BWFCGUIXT, 2024-04-19:
Seems that shorting GND + Reset till it's done configuring helps it configure.

[image]

But after initial configuration no messages are received anymore.

BWFCGUIXT, 2024-04-19:
So seems we installed the wrong update, I can see a Commit from yesterday deleting the update and replacing it.

[image]

Hope we can somehow flash the correct update.

BWFCGUIXT, 2024-04-19:
Seems the update file we used was actually for the BZ firmware.

pvvx, 2024-04-20:
Version 0.1.2.1 is not a release yet
And at the moment version 0.1.2.1 is not yet complete. There is no converter for Z2M yet. Not included in the update lists...
Due to the rush, the firmware for [BZdevice ](https://github.com/pvvx/BZdevice) was recorded... Zigbee OTA and BLE OTA work for this version.

pvvx, 2024-04-20:
> Same here with 2 devices, I hope we can do OTA somehow and not USB flashing 

Using Zigbee OTA, update the firmware [in the BLE option](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota) and write any version to TelinkMiFlasher.html.

BWFCGUIXT, 2024-05-05:
@TeHashX  Thank you for the instructions 👍 

Managed to fix all 5!

closed 2025-11-07 by pvvx