pvvx/ZigbeeTLc #86: Что за BLE ATCz BTHome устройство на прошивке Z03MMC_v0121?

vitkov83, 2024-04-18:
Ожидал Zigbee, а получил BLE, я что-то перепутал?

pvvx, 2024-04-18:
What file did You use for the firmware?

vitkov83, 2024-04-18:
> What file did You use for the firmware?

https://github.com/pvvx/ZigbeeTLc/blob/master/bin/Z03MMC_v0121.bin

pvvx, 2024-04-18:
Binaries version 0.1.2.1 have been fixed.

The previous files were incorrect, created from a different repository. A full update to v0.1.2.1 has not yet been made.

vitkov83, 2024-04-18:
> Исправлены двоичные файлы версии 0.1.2.1.
> 
> Предыдущие файлы были неверными и созданы из другого репозитория. Полное обновление до версии 0.1.2.1 еще не сделано. Конвертеров для Z2M для версии 0.1.2.1 пока нет. Тот, который вы скачали, является гибридом с https://github.com/pvvx/BZdevice . Также еще не обновлен "firmware.json" для TelinkMiFlasher.html, так как не все в версии 0.1.2.1 проверено...

Понял, спасибо! Делаю откат на v0.1.2.0

vitkov83, 2024-04-18:
Понял, спасибо! Делаю откат на v0.1.2.0

closed 2024-04-18 by vitkov83

pvvx, 2024-04-18:
The downloaded file has version 0.2.0.5.
I don’t know how to downgrade the version in Z2M, since I don’t use Z2M.
But you can always roll back via BLE firmware.

vitkov83, 2024-04-18:
> The downloaded file has version 0.2.0.5. I don’t know how to downgrade the version in Z2M, since I don’t use Z2M. But you can always roll back via BLE firmware.

В Z2M всё нормально, обновление не предлагает, я просто глянул репозиторий и увидел "новую версию 121",  прошивал через USB-COM, решил глянуть ради любопытства )