pvvx/ZigbeeTLc #88: Flashing LYWSD03MMC with USB-UART - the right way

TeHashX, 2024-04-22:
Hi, I updated 2 devices with wrong firmware through OTA and are in constantly bootloop https://github.com/pvvx/ZigbeeTLc/issues/85
Tried to flash with usb-uart web tool https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html but now the screen is blank.
My adapters are not supported? The used schematic is wrong? Tried swapping RX with TX, same thing.
[image]
I also tried with Voltlink adapter without success.
[image]
I'm not sure if it's the right place for asking but please advice.
Thank you!

pvvx, 2024-04-23:
If with a resistor between RX and TX then it's [TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py)

https://github.com/pvvx/TlsrComSwireWriter

There is an example here:
[Flashing firmware with USB to UART](https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-firmware-with-usb-to-uart)

TeHashX, 2024-04-23:
Thank you! With TlsrComSwireWriter I recovered both devices 👍
The schematics was ok but not the web flasher.

pvvx, 2024-04-23:
Option 2:

[image]

https://github.com/pvvx/TlsrComSwireWriter

File "https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py"

There is an example here:
[Flashing firmware with USB to UART](https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-firmware-with-usb-to-uart)