pvvx/ZigbeeTLc #91: TS0201 invalid temperature and humidity

yeahme49, 2024-05-07:
I flashed my TS0201 to 0.1.2.1 but I'm not getting valid temperature or humidity readings once it's paired to ZHA.

[image]

Any ideas? Thanks

pvvx, 2024-05-08:
[image]

shing6326, 2024-05-30:
my device also not working after installed the firmware,  my device id is "_TZ3000_rdhukkmi", may be the sensor was changed?

below is the z2m status from the device
{
    "battery": 100,
    "humidity_calibration": 0,
    "last_seen": "2024-05-30T19:16:57.416Z",
    "linkquality": 112,
    "measurement_interval": 10,
    "temperature": -327.68,
    "temperature_calibration": 0,
    "update": {
        "installed_version": 18952193,
        "latest_version": 18952193,
        "state": "idle"
    },
    "voltage": 3100,
    "humidity": 655.35,
    "update_available": null
}

Steamerzone, 2024-08-19:
I've got the same problem on firmware v1.22, reconfiguring the device is failing also.

          "0x0402": {
            "endpoint_attribute": "temperature",
            "attributes": {
              "0xfffd": {
                "attribute_name": "cluster_revision",
                "value": 1
              },
              "0x0000": {
                "attribute_name": "measured_value",
                "value": -32768
              }
            },
            "unsupported_attributes": {}
          },
          "0x0405": {
            "endpoint_attribute": "humidity",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 65535
              }
            },
            "unsupported_attributes": {}
          }
        },

Steamerzone, 2024-08-19:
I got it working again by flashing the ZTS0201Z3000_v0122.bin file with a serial programmer.

The temperature and humidity values work fine now.
Only in HomeAssistant the reconfigure option still doesn't work:

[image]

```
2024-08-19 20:03:44.867 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0001]: Performing cluster binding
2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x24 request
2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 867553, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=36, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'$\xe9\x1c\xf62l8\xc1\xa4\x01\x01\x00\x03\xdb\xc0\xe8)\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2024-08-19 20:03:44.867 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0402]: Performing cluster binding
2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x25 request
2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 867912, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=37, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'%\xe9\x1c\xf62l8\xc1\xa4\x01\x02\x04\x03\xdb\xc0\xe8)\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2024-08-19 20:03:44.868 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0020]: Performing cluster binding
2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x26 request
2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 868255, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=38, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'&\xe9\x1c\xf62l8\xc1\xa4\x01 \x00\x03\xdb\xc0\xe8)\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2024-08-19 20:03:44.868 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0204]: Performing cluster binding
2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x27 request
2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 868613, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=39, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b"'\xe9\x1c\xf62l8\xc1\xa4\x01\x04\x02\x03\xdb\xc0\xe8)\x00K\x12\x00\x01"], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
```

sebgus, 2024-10-17:
> I got it working again by flashing the ZTS0201Z3000_v0122.bin file with a serial programmer instead of OTA.
> 
> The temperature and humidity values work fine now. Only in HomeAssistant the reconfigure option still doesn't work:
> 
> ![Screenshot 2024-08-19 195859](https://private-user-images.githubusercontent.com/22254368/359198643-eb83a588-2095-4292-88ab-b45cdf71ce99.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkxNDM4MjksIm5iZiI6MTcyOTE0MzUyOSwicGF0aCI6Ii8yMjI1NDM2OC8zNTkxOTg2NDMtZWI4M2E1ODgtMjA5NS00MjkyLTg4YWItYjQ1Y2RmNzFjZTk5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE3VDA1Mzg0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRiOTk4ZDNlYzllNGE2Yzc5NmM0YjM5NTFlNmUwNWQ0ZDE0YmMwMTJiMzhhMjQ1ZTVmYWI3MmJkN2U0NzU3OTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gaFJt3AcBBVnwXeKaBqyYOhPifxZ_Gp_D4crpkpe_mg)
> 
> ```
> 2024-08-19 20:03:44.867 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0001]: Performing cluster binding
> 2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x24 request
> 2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 867553, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=36, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'$\xe9\x1c\xf62l8\xc1\xa4\x01\x01\x00\x03\xdb\xc0\xe8)\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
> 2024-08-19 20:03:44.867 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0402]: Performing cluster binding
> 2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x25 request
> 2024-08-19 20:03:44.867 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 867912, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=37, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'%\xe9\x1c\xf62l8\xc1\xa4\x01\x02\x04\x03\xdb\xc0\xe8)\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
> 2024-08-19 20:03:44.868 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0020]: Performing cluster binding
> 2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x26 request
> 2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 868255, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=38, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'&\xe9\x1c\xf62l8\xc1\xa4\x01 \x00\x03\xdb\xc0\xe8)\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
> 2024-08-19 20:03:44.868 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xDCC5:1:0x0204]: Performing cluster binding
> 2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy.device] [0xdcc5] Extending timeout for 0x27 request
> 2024-08-19 20:03:44.868 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2024, 8, 19, 18, 3, 44, 868613, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDCC5), dst_ep=0, source_route=None, extended_timeout=True, tsn=39, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b"'\xe9\x1c\xf62l8\xc1\xa4\x01\x04\x02\x03\xdb\xc0\xe8)\x00K\x12\x00\x01"], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
> ```
How did you flash it with a serial programmer and which one did you use? My TS0201 also behaves strange after an OTA update and I can't seem to bring it to life again.

vulpes2, 2025-05-22:
If you have a `_TZ3000_rdhukkmi` then the TS0201_TZ3000 firmware is not going to work. This variant has an SHT30 instead of a CHT8305, see #201 for details.

vulpes2, 2025-05-23:
SHT30 and a few other sensors have been enabled in the `ZTS0201Z3000` firmware, my `_TZ3000_rdhukkmi` is working now: https://github.com/pvvx/ZigbeeTLc/commit/a74aeee11def0bbf840888727b5fc5e6d3453bad