pvvx/ZigbeeTLc #94: TS0201 by _TZ3000_0s1izerx - No OTA image is available in ZHA

hippich, 2024-05-15:
I have TS0201 temperature/humidity sensor, but it is round one, not rectangular. It reports:

TS0201 by _TZ3000_0s1izerx
IEEE: a4:c1:38:b4:55:3f:82:f9
manufacturer_code: 4417
image_type: 54179
current_file_version: 70

While TS0201 firmware in this repo has `"imageType": 529,`, which is, i guess, the reason ZHA refuses to upload firmware.

Question - are these actually different sensors and forcing ZHA to upload it will brick it, or it is just all the same with just different image_type?

For the reference - this is the type of sensor I have :

[image]

pvvx, 2024-05-15:
manufacturer_code: 4417 -> 0x1141
image_type: 54179 -> 0xDAD3
current_file_version: 70 -> 0x0046 

OTA file: 1141-d3a3-01983001-TS0201z_v0121.zigbee

https://github.com/pvvx/ZigbeeTLc/tree/master/bin

hippich, 2024-05-16:
Thanks! ZHA picked it up!

hippich, 2024-05-16:
And I think it semi-bricked itself after the update, so just FYI if anyone will be looking into doing it. Will keep tinkering with it.

symptoms:

when pressing button, red led lights up, but very dim. when holding for 7 seconds to re-pair, red led lights up solid, but again dim, and after a few seconds turns off. ZHA can't find it so far.

hippich, 2024-05-16:
Opened it up. Looks like there is no ISP header. I wonder if I can connect directly to ZTU board instead to flash it with USB-TTL programmer?

[image]

pvvx, 2024-05-16:
> when pressing button, red led lights up, but very dim. when holding for 7 seconds to re-pair, red led lights up solid, but again dim, and after a few seconds turns off. ZHA can't find it so far.

This is a normal reaction: reset the settings and pairing.
 
> It does (dimly) blinks every few seconds. So I guess it does something.

Indication that there is no pairing.

After pairing, it flashes quickly several times.

pvvx, 2024-05-16:
> Opened it up. Looks like there is no ISP header. I wonder if I can connect directly to ZTU board instead to flash it with USB-TTL programmer?

https://pvvx.github.io/TH03Z/