pvvx/ZigbeeTLc #95: MHO-C401 Webupdater Bricks one of two device

Wolfscowl, 2024-05-15:
Hi, I bought two MHO-C401 and tried to flash the zigbee FW.

**Device A (left on the photos)**
- Packaging date; 2023 05
- Board date: 2021.03
- Everything worked without any problems

**Device B (right on the photos)**
- Packaging date 2024 01
- Board date: 2021.03
- There is no Zigbee firmware to choose from in the webupdater. Only the custom FW
- I have flashed the custom  firmware as a test
- The device can then be reconnected, but the e-paper display always shows the same values.
- In the WebUpdater, however, it reads current values.
- I tried to reflash the original firmware, but the epaper is still dead

Both devices should be the new version, right?
What is different about device B? Why is there no custom FW and why is the display now dead?
Sorry for the bad photos. My cell phone camera is miserable.

[image]
[image]
[image]

Wolfscowl, 2024-05-17:
Thanks for your reply.
Then the second device must be an Azarton, but the packaging looks completely identical.

Gekkon-76, 2025-04-11:
2024.01 The screen also froze after reflashing. How did you solve the problem?

pvvx, 2025-04-11:
> 2024.01 The screen also froze after reflashing. How did you solve the problem?

The solution is simple: update the firmware via OTA in TelinkMiFlasher to another HW version: MHO-C401 <-> MHO-C401N.

MHO-C401: https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/bin/MHO_C401_v51.bin
MHO-C401N: https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/bin/MHO_C401N_v51.bin

Paste this link into the OTA request file.