pvvx/ZigbeeTLc #96: LYWSD03MMC-z 0.1.2.1 When changing interface parameters, a reboot is required!

dodgerz2000, 2024-05-16:
When changing interface parameters, a reboot is required! When disabling smile (Attr: 0x0002), it does not disappear until reboot. When the display is turned off (Attr: 0x0106) - the display does not turn off, the image is frozen (not updated).

pvvx, 2024-05-16:
https://github.com/pvvx/ZigbeeTLc/issues/30#issue-2034396126

* Not on all HW variants, turning off the display leads to a significant reduction in consumption. Better results are obtained by increasing the measurement interval.

pvvx, 2024-05-16:
> When disabling smile (Attr: 0x0002), it does not disappear until reboot. When the display is turned off (Attr: 0x0106) - the display does not turn off, the image is frozen (not updated).

This type of equipment requires the battery to be temporarily disconnected when the display off option is enabled.

Meisam-kh, 2024-05-17:
Hi
Would you please tel me in general which below HW versions of xiaomi LYWSD03MMC is better to use for firmware update with ZigbeeTLc ?
HW:B1.4 or HW:B1.7 ?