pvvx/ZigbeeTLc #98: Temperature not updating, but humidity does

luvxinh, 2024-05-30:
[image]

[image]

[image]

So in my case, the humidity reading is updated fine but the temperature reading does not update, unless I go into Manage zigbee device and force update, then it would have the new reading but then again, no update from then.

pvvx, 2024-05-30:
[image]

luvxinh, 2024-05-30:
I got this.
[image]

Also this is in the log.

Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/__init__.py:560
integration: Zigbee Home Automation (documentation, issues)
First occurred: 11:54:58 (2 occurrences)
Last logged: 11:54:58

[0xAF30:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

I hope this can be of help

pvvx, 2024-05-30:
>  [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

I don't have such a message.
The coordinator on EFR32MG21 is used.
[image]

luvxinh, 2024-06-10:
I'm marking this as closed because after some time, the sensor ... updates. Frequency wise, it's haywire though ...

Still I appreciate the zigbee patch

closed 2024-06-10 by luvxinh

luvxinh, 2024-08-20:
Hi

Just want to clear the confusion for anyone reading ... it was **not** because the sensor did not update, but rather the temperature did not change for more than 0.5 degree celcius ... took my awhile to realize that.

Thanks for the firmware.

pvvx, 2024-08-20:
If there has been no change for 0.5C (1%), ZHA sets the update time to 900 seconds by default.