pvvx/ZigbeeTLc #100: Invalid value for comfort_temperature_min/_max

TheGroundZero, 2024-06-03:
I recently started getting this error message in Home Assistant.

```
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:195
integration: MQTT (documentation, issues)
First occurred: 15:01:10 (16 occurrences)
Last logged: 15:06:09

Invalid value for number.temphum_xander_comfort_temperature_max: 250 (range -50.0 - 120.0)
Invalid value for number.temphum_mbr_comfort_temperature_min: 200 (range -50.0 - 120.0)
Invalid value for number.temphum_mbr_comfort_temperature_max: 250 (range -50.0 - 120.0)
Invalid value for number.temphum_badkamer_comfort_temperature_min: 200 (range -50.0 - 120.0)
Invalid value for number.temphum_badkamer_comfort_temperature_max: 250 (range -50.0 - 120.0)
```

[image]

robvanoostenrijk, 2024-07-09:
Reported in https://github.com/pvvx/ZigbeeTLc/issues/106.

You need to update the device to firmware v0.1.2.2 and the latest converter fix in zigbee-herdsman-converters from Zigbee2MQTT.

TheGroundZero, 2024-07-09:
> You need to update the device to firmware v0.1.2.2 and the latest converter fix in zigbee-herdsman-converters from Zigbee2MQTT.

Do I need to add a custom converter or is this now included in Z2M?

robvanoostenrijk, 2024-07-09:
The changes for the comfort settings scaling were just submitted to Z2M upstream and should be available in the next build.

Note that the external convertor was already integrated in Z2M before and you no longer require it external convertor for pvvx devices to work (Firmware v0.1.2.1+ only).

kuchar91, 2024-07-21:
Where is firmware v0.1.2.2? I can see only 0.1.2.1

robvanoostenrijk, 2024-07-21:
bin folder in repo has the v0.1.2.2 versions:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin